barbeau/gpstest #277: Add Blackout Dark mode

barbeau, 2019-04-19:
**Summary:** 

From user:

>please add a real black theme for amoled. and grey theme isn't dark.

I'd like this too.  There are some dark theme options for later versions of Android that we aren't currently using that we can take a look at:
https://www.xda-developers.com/android-q-dark-mode-overview/

**Steps to reproduce:** 

Go to settings and enable "Dark mode"

**Expected behavior:** 

Give me an option for blackout mode option for true black background, or make dark mode darker

**Observed behavior:** 

Dark mode has dark gray background

**Device and Android version:** 

N/A

mickhayes, 2019-08-02:
Looking at gpstest Settings (3.3.4) "Use Dark Theme" 

Black is 0xff000000 in Alpha-R-G-B - https://developer.android.com/reference/android/graphics/Color

In GPSTest\src\main\res\values\colors.xml:

<color name="navdrawer_background_dark">@color/cardview_dark_background</color>

This has Alpha-R-G-B value: FF 20 20 20, i.e. not totally black.

barbeau, 2019-08-02:
@mickhayes The current theming in GPSTest is a mix of legacy Android themeing and manually adjusting colors when the legacy theme didn't work for whatever reason.  So unfortunately it's not going to be as simple as just changing `cardview_dark_background`.  I'm hopeful that the new system theme support for dark mode in Android (see above links in initial post) will simplify a lot of this code, although I haven't had a chance to deep-dive on this myself.

barbeau, 2019-08-02:
Here's the original commit that adding support for the dark theme, although the implementation has gotten more complex than with later additions to Sky and Accuracy views, etc.:
https://github.com/barbeau/gpstest/commit/3939beb447dba06971cec70e04b363307f5fefe5

But you can see the use of the system `AppTheme.Dark.NoActionBar` theme there.

barbeau, 2019-08-14:
Google I/O 2019 app code is now live, which includes dark theme implementation:
https://9to5google.com/2019/08/14/io-2019-open-source/amp/

barbeau, 2019-09-27:
See Dark Theme in material components:
https://material.io/develop/android/theming/dark/

GeoZac, 2020-07-26:
Maybe a explanation not to adopt the darkest black and stick to the Material Themes :
https://www.xda-developers.com/amoled-black-vs-gray-dark-mode/

sm4rk0, 2020-07-30:
That article makes sense, but
1. This app has #303030 for "dark" background (Although a comment above says #202020. I took a screenshot and opened it in Simple Draw app, then went to color chooser and picked the same color), which is far brighter than #121212 from the article
2. Some people just prefer black over (dark) gray, regardless of battery savings

barbeau, 2020-08-06:
Hopefully switching to the official dark mode will also fix this issue in Settings on legacy devices:

>In the dark environment with dark mode I can't see what option, it is or not selected! 
>
>App version: v3.6.4 (18066-google)
Model: E39
Android version: 4.4.2 / 19

[image]

barbeau, 2021-11-05:
This should include the settings for:
* Dark mode
* Light mode
* System default

barbeau, 2023-10-21:
Not stale

barbeau, 2024-05-09:
Not stale