barbeau/gpstest #279: Color issue: Same color for font and BG in pop-up dialogue

uli-on, 2019-04-20:
If GPS is deactivated, the app puts up a dialogue, asking whether GPS shall be activated now or left as is. The two blue buttons come with the same color for font and background and can only be dustinguished by holding the button pressed as long as it tskes for the colors to change.

GPSTest 3.3.1
Android 6.0.1
Motorola Moto G4

barbeau, 2019-04-20:
@uli-on Thanks for the report!  Could you please add screenshots?

mickhayes, 2019-07-25:
This is not a problem and working fine (text of both "NO" and "ENABLE GPS" buttons is white) in this environment:

App version: v3.3.1 (47-google)
Model: Samsung Galaxy A5 (2017) SM-A520F.
Android version: 8.0.0
Google Play Services App: 17.7.85

goldfndr, 2019-07-27:
It remains unsolved for me (text of NO and ENABLE GPS buttons is blue, identical to background, for both light theme and dark theme) in these environments:

App version: v3.3.4
Model: Samsung Galaxy Note4 SM-N910T
Android version: 6.0.1 / 23 (stock, not rooted)
Google Play Services App: 18.3.81

App version: v3.3.4
Model: Samsung Galaxy S5 SM-G900P
Android version: 6.0.1 / 23 (stock, not rooted)
Google Play Services App: 17.7.85

mickhayes, 2019-07-27:
Fails on:
App version: v3.3.4
Model: Samsung Galaxy S5 SM-G903F
Android version: 6.0.1
Google Play Services App: 18.3.81

jayhayes, 2019-07-27:
Works fine on:
App version: v3.3.4
Model: Samsung Galaxy J5 (2016)
Android version: 7.1.1
Google Play Services App: 18.3.81

mickhayes, 2019-07-27:
Looking at:

    <string name="enable_gps_message">GPS is disabled. Enable it now?</string>
    <string name="enable_gps_positive_button">Enable GPS</string>
    <string name="enable_gps_negative_button">No</string>
in GPSTest/src/main/res/values/strings.xml and other lang alternatives.

enable_gps_positive_button used at: GPSTest/src/main/java/com/android/gpstest/GpsTestActivity.java thusly:

    /**
     * Ask the user if they want to enable GPS
     */
    private void promptEnableGps() {
        new AlertDialog.Builder(this)
                .setMessage(getString(R.string.enable_gps_message))
                .setPositiveButton(getString(R.string.enable_gps_positive_button),
                        new DialogInterface.OnClickListener() {

                            public void onClick(DialogInterface dialog, int which) {
                                Intent intent = new Intent(
                                        Settings.ACTION_LOCATION_SOURCE_SETTINGS);
                                startActivity(intent);
                            }
                        }
                )
                .setNegativeButton(getString(R.string.enable_gps_negative_button),
                        new DialogInterface.OnClickListener() {

                            public void onClick(DialogInterface dialog, int which) {
                            }
                        }
                )
                .show();
    }

barbeau, 2019-07-28:
My guess is that the root cause of this issue is the Android X support libraries not applying theme colors correctly on Android 6.x. Would be worth some Goolging to see if others have encountered this.

barbeau, 2019-07-31:
Actually, looking at this more, this dialog actually *isn't* using the support library (it's using `AlertDialog.Builder` and not `androidx.appcompat.app.AlertDialog.Builder`), hence the issue.  I'll push a fix shortly.

closed 2019-07-31 by barbeau

goldfndr, 2019-07-31:
Is it necessary to specify `androidx.appcompat.app.AlertDialog` rather than `AlertDialog`? The other files don't.

barbeau, 2019-07-31:
The other files import `androidx.appcompat.app.AlertDialog` at the file of the file, so it's fine to use just `AlertDialog` inline.  For some reason that I don't recall I had the fully qualified name `androidx.appcompat.app.AlertDialog` inline in GpsTestActivity - it can be changed to match the other files so instead of always using the fully-qualified name inline it's imported.  I'll change this so it matches.  The main issue is that we don't want the import at the top of the file of `android.app.AlertDialog`, which caused the problem here because it's not the support library.

commit 2f6806db (2019-07-31): Fix #279 - Use support library for dialog to enable GPS

Also convert to lambdas for this dialog