barbeau/gpstest #286: Italian translation

Giolly90, 2019-04-24:
I just completed Italian translation on Transifex, now what's next?
Can I check the result and optimize?

barbeau, 2019-04-30:
@Giolly90 Thanks for doing this!  Yes, I'll produce a new build shortly that you can test.

barbeau, 2019-04-30:
@Giolly90 Could you please uninstall the current version on your device and install this version and test?

https://www.dropbox.com/s/6e95qg800jjx94r/GPSTest-google-debug.apk?dl=0

barbeau, 2019-04-30:
Sorry, please try this one instead:
https://www.dropbox.com/s/abry1x44hz7eufh/GPSTest-google-debug.apk?dl=0

Giolly90, 2019-04-30:
Thank you! I installed the new version and I am very happy with the result!
I looked through all the menus and settings and now I would like to fix some little problems and improve some translation.
I will work on this starting from 6th May, I'll let you know when I'm done asap

Giolly90, 2019-05-09:
Hi Sean, I made the changes, can you please send me a new version of the app?

I have also some questions to make a better translation:
- What are Assist data, XTRA data and Time data?
- What is callback status?
- Can you please make an example of Diluition of Precision?
- Can you please explain better how Precsion tab works?

Can you add me in the Acknowledgments as contributor for Italian translation?

Thanks!

barbeau, 2019-05-09:
@Giolly90 Thanks!  Here's a new version:
https://www.dropbox.com/s/abry1x44hz7eufh/GPSTest-google-debug.apk?dl=0

Also note that in this version you can change the language in Settings, if that helps.

>What are Assist data, XTRA data and Time data?

See https://github.com/barbeau/gpstest/blob/master/FAQ.md#what-are-the-menu-buttons-for

>What is callback status?

This is more of an error code that the user really isn't intended to understand, but they could tell me this and it would help me figure out a problem.  99% of users will never see this, so you can leave it untranslated if you want.   A "status callback" refers to the value that the device provides about the state of the GNSS.

>Can you please make an example of Diluition of Precision?

This is a metric used to express how the position of the satellites in the sky can affect the quality of the location that is calculated by the device.  For more see https://en.wikipedia.org/wiki/Dilution_of_precision_(navigation).

>Can you please explain better how Precsion tab works?

This is "Accuracy", not "Precision".  It's used to measure how accurate the device's position is.  The user can tap on the map to tell GPSTest where they are actually located in real life.  Then, the app monitors the calculated GNSS location and determine the difference between the actual position of the device and the calculated position - this difference is the accuracy.

I added you as "Giolly90" in the English strings!

barbeau, 2019-05-09:
Quick note - some of the strings on the top of "Status" screen ("Stato") need to be shortened - even on a wide Galaxy S8+ they are pushing values off-screen:

[image]

Giolly90, 2019-05-09:
When I install new app from the link you sent, I don't see the last changes made in Transifex. Can you please doublecheck?

The screen you've posted is infact from the old translation, maybe now I've fixed the "space" iasue

barbeau, 2019-05-09:
Thanks for letting me know!  For some reason the tool doesn't always seem to pull the latest unless I force it.  This one should be updated:
https://www.dropbox.com/s/z7q9jcmdqp6pp87/GPSTest-google-debug.apk?dl=0

Closer, but still some cutoff:
[image]

Giolly90, 2019-05-10:
Hope the last changes will fix all the cutoff, please share the new version and if you can also the Status screenshot from your device (with my device everything was fine also with the old release)

Thanks!

barbeau, 2019-05-14:
@Giolly90 Most recent translation for Status looks good on Galaxy S8+, thanks!

[image]

On the Accuracy screen, would it be possible to reduce the summary text to one line so that the bottom of the button isn't cut off?

[image]

This APK should have the latest translation:
https://www.dropbox.com/s/vmug0ixkavm375u/GPSTest-Russian.apk?dl=0

I'm going to do another beta release as well.

Giolly90, 2019-05-15:
All done! Hopefully this is the def. version for Italian translation.
As always, please send me an updated relase to test a bit

goldfndr, 2019-05-15:
I'm curious about the "m s.l.m." translation for Mean Sea Level. Shouldn't it be more like "livello medio del mare"? Alteratively, what is the Italian expansion (words) of "m s.l.m."?

Giolly90, 2019-05-15:
> I'm curious about the "m s.l.m." translation for Mean Sea Level. Shouldn't it be more like "livello medio del mare"? Alteratively, what is the Italian expansion (words) of "m s.l.m."?

I started from the diction: Metres above sea level, in Italian: Metri Sul Livello del Mare (m s.l.m.)

barbeau, 2019-05-15:
@Giolly90 New translation looks good, thanks!

[image]

I'm going to close this issue and will work on pushing out a new beta with the changes.

closed 2019-05-15 by barbeau

goldfndr, 2019-05-16:
@Giolly90 I suspected that. In that case, the "m" at the start is redundant and possibly inaccurate (units are shown after the number and can be changed to feet) and perhaps just "s.l.m." should be shown.

Giolly90, 2019-05-16:
Got your point @goldfndr ! And yes, you're right, I've made the change on Transifex leaving only "s.l.m." and also changed the Help section consequently.
Thank you!

Giolly90, 2019-06-26:
Hi Sean, I just made a little fix within the Italian translation, can you force the update to have it shown in the app? Or do we have to wait for the next release?

Thanks!

barbeau, 2019-07-02:
@Giolly90 Unfortunately all updates to translations require a new app release.  I'm working on that now, though, so the change will be visible in the next release.