barbeau/gpstest #287: AppCrash since v3.3.0 [LogCat]

github-tomster, 2019-04-26:
**Summary:** 

app since version 3.3.0 crashes on appstart

**Steps to reproduce:** 

open drawer, tap gpstest icon, toast appears: gpstest died

**Expected behavior:** 

to work like version 3.2.10

**Observed behavior:** 

it crashes...

**Device and Android version:** 

samsung galaxy note 3 
android 4.4.2 stock
f-droid installation

**LogFile attached**
[gpstest.txt](https://github.com/barbeau/gpstest/files/3121389/gpstest.txt)

barbeau, 2019-04-27:
Thanks for reporting this! This is an earlier version of Android so it's going to be a little harder for me to reproduce, as it doesn't seem to happen on later Android versions.

Do you still see the crash if you uninstall and reinstall the new version of the app?

github-tomster, 2019-04-28:
yes, i tried to uninstall and start clean with the same result of the crash.
only downgrading helps.
i wish you the best success to find the right code 👍

barbeau, 2019-04-30:
@github-tomster Are you installing the new version of GPSTest directly from F-Droid, or are you downloading and installing the APK from GitHub or somewhere else?

github-tomster, 2019-04-30:
@barbeau 
f-droid, indeed that is correct.

barbeau, 2019-05-03:
So it looks like this is an issue in the underlying Google ConstrainLayout library I'm using for MotionLayout in the alpha2 version:
https://www.reddit.com/r/androiddev/comments/b9nqxm/constraintlayout_200_alpha_4/ekhovyg/

I've tried alpha5 version of the library although that has issues with the TextViews for setting latitude/longitude/altitude in Accuracy view not laying out correctly:
https://github.com/barbeau/gpstest/tree/motion-layout-alpha5

I'll keep looking for a solution...

Schyrsivochter, 2019-05-06:
Confirming, this issue is present on Android 4.1.2, too.

barbeau, 2019-05-20:
From some testing it looks like the beta1 release has fixed some of these issues, although it requires some re-work for the new version.  I'll try pushing out a new beta with this and hopefully it should solve the issue.

github-tomster, 2019-05-20:
we will see.
thank you for looking into it.

barbeau, 2019-05-22:
@github-tomster and @Schyrsivochter v3.3.2 is now rolling out that contains this fix - could you please give it a try and see if it resolves the crash?

cvzi, 2019-05-25:
v3.3.2 works for me again on android 4.4.2. Thanks for the great work!

barbeau, 2019-05-25:
Great!

github-tomster, 2019-06-03:
i can also confirm: 3.3.2 works again. thank you!

Tony-il-Capo, 2019-06-11:
Hi
I'm sorry, but I still have this problem on my old Xperia S, app crash right at start, logcat is similar to the above
[logcat](https://github.com/barbeau/gpstest/files/3274327/logcat.txt)

Last version that works on this device is 3.2.10

barbeau, 2019-06-11:
@Tony-il-Capo What Android version is this device running, and is it the original OEM version or a custom ROM?

Also, can you please try the following steps with both of these APKs and let me know if you see the crash on both versions?

* https://github.com/barbeau/gpstest/releases/download/v3.3.2/googleRelease-v3.3.2.apk
* https://github.com/barbeau/gpstest/releases/download/v3.3.2/osmdroidRelease-v3.3.2.apk

1. Uninstall the current GPSTest app(s) on your device
1. Install the above linked APK

Tony-il-Capo, 2019-06-15:
Sorry for the late reply, too busy...
Unfortunately they both crash, I had tested them before and to make sure just retested them now, both crash at start.

barbeau, 2019-06-15:
@Tony-il-Capo What Android version is this device running, and is it the original OEM version or a custom ROM?

Tony-il-Capo, 2019-06-15:
Sorry, I forgot, I was just editing comment... anyway...

Device runs Android 4.1.2 with just 'minor' tweaks.
As I said before v 3.2.10 runs OK.

barbeau, 2019-06-15:
@Tony-il-Capo What are "minor tweaks"?

Tony-il-Capo, 2019-06-16:
Little things, like no background process (low priority/importance) allowed, removed gapps, removed google services, init.d added, force GPU composition... iptables firewall script... obviously  root access, can't remember anything more.

Most of those should not be an issue, only, maybe if an app depends on google services...

barbeau, 2019-06-17:
@Tony-il-Capo I'm going to re-open this as it's likely related to comparability issues with the Google ConstraintLayout library I'm using for MotionLayout for Android 4.1.x (API level 16).  It's possible Google fixed the compatibility issue with Android 4.4.x (API level 19) but not the older version of Android.  In other words, fixing this issue will required Google fixing the bug in ConstraintLayout library in a new release of that library.

barbeau, 2019-06-17:
@Tony-il-Capo Also FYI, you can always download old releases as a workaround for now from https://github.com/barbeau/gpstest/releases.

Tony-il-Capo, 2019-06-18:
Thanks.

> In other words, fixing this issue will required Google fixing the bug in ConstraintLayout library in a new release of that library.

So, unlikely to happen? If so, will you drop compatibility for older android versions?

barbeau, 2019-06-18:
@Tony-il-Capo I wouldn't saw unlikely if the issue was already fixed for Android 4.4, but just an unknown timeframe. I would only drop compatibility for older Android versions if the library forced me to do that.

On this note, the original [introduction for MotionLayout from Google](https://medium.com/google-developers/introduction-to-motionlayout-part-i-29208674b10d) said the following:

>Finally, as a part of ConstraintLayout 2.0, it is available as a support library, backwards compatible to API level 14 (ICS) : which means it supports 99.8% of current Android devices (see distribution chart).

API level 14 is Android 4.0, which is the `minSdkVersion` that the latest release of GPSTest supports.

However, looking at the [MotionLayout reference documentation](https://developer.android.com/reference/android/support/constraint/motion/MotionLayout) now, it says:

>Note: MotionLayout is available as a support library that you can use on Android systems starting with API level 18 (JellyBean MR2).

I'm going to see if I can hunt down more info on which of these statements is true.

jyhpsycho, 2019-07-29:
I have Samsung Galaxy S2 HD LTE(Android 4.1.2, the latest official firmware). The latest gpstest version can use with that is 3.2.10. It seems that the MotionLayout is applied from 3.3.0(It just a guess because I'm not an android developer, and the release note does not say that). I don't know which part of this app using MotionLayout, but it seems that it may be replaced(I don't know what the module provides similar functionality) or just dropped if you wish to support these older devices afterward. Or, minimum API level should be set to 18 according to current official [MotionLayout reference documentation](https://developer.android.com/reference/android/support/constraint/motion/MotionLayout) you mentioned.

barbeau, 2019-07-31:
I just tested the Google Constraint Layout v2.0.0-beta2 release (https://github.com/barbeau/gpstest/tree/constraintlayout-2-0-0-beta2), and I'm still seeing the crash on API level 16.  So, at this point, given the reference docs for MotionLayout and this release, I'm not confident that Google will fix this issue in the support library.

So yes, at this point v3.3 and higher of GPSTest will need to be restricted to only API Level 18 and up - in other words, for new releases of GPSTest I'll need to drop support for API Level 17 and lower.  I'll work on pulling together a hotfix for v3.2.x that will allow users with devices API Level 17 and lower to continue using that version without a crash. Unfortunately, though, this means that the Accuracy feature will only be available to users with devices API Level 18 and up.

barbeau, 2019-08-07:
I'm re-releasing v3.2.10 as v3.2.11 with a higher versionCode than the latest version (v3.3.4) so devices <= 17 pick up this update instead of v3.3.4, which should fix this issue. I'm also releasing a new v3.4.0 for API levels 18 and up with a few new features and bug fixes.

closed 2019-08-07 by barbeau

phm1015, 2019-12-17:
i also found the same issue in api 16
got any solution for api 16??

barbeau, 2019-12-17:
@phm1015 Unfortunately no, I ended up having to drop support for API <=17 after adopting MotionLayout.

commit d93ebfc1 (2019-05-20): Fix #287 - Update to ConstrainLayout library version beta1

* Also bump lifecycle-extensions library to latest beta
commit 7dd6b9cc (2019-08-07): Fix #287 - Bump minSdkVersion to 18

v3.2.10 will be separately re-released as v3.2.11 to fix crashes on API 14-17 devices. See https://github.com/barbeau/gpstest/releases/tag/v3.2.11.