barbeau/gpstest #288: Skyview exceeds screen height in landscape orientation

lbdroid, 2019-04-30:
Its clearly using the same layout as in portrait mode, where it is sizing the skyview to fit the width on the screen. In the case of landscape orientation, the width is greater than the height, which means that the view is cut off unless you scroll.

In landscape mode, it should have a different layout, with the legend perhaps *beside* the skyview instead of below it.

barbeau, 2019-04-30:
@lbdroid Thanks for the feedback!  I agree that this could be improved.

goldfndr, 2019-05-03:
As more satellites become usable, having this large layout available can help graphically distinguish particular satellites that are close to the same elevation+azimuth. [GNSS View](https://play.google.com/store/apps/details?id=com.nec.android.qzss.gnssview) can be used on Android devices that lack specific constellation access to show what this crowded view currently looks like.

closed 2022-04-11 by stale[bot]