barbeau/gpstest #294: Shorten Czech translation

barbeau, 2019-05-14:
**Summary:** 

@LLZN on the Czech translation, after you get a fix the top right column on the Status page hangs off the edge of the screen - see below on a Samsung Galaxy S8+:

[image]

Also, on the Accuracy screen, could you try to reduce the summary text to one line so that it doesn't push the button off the bottom of the card?

[image]

Could you please try and shorten some of these strings using Transifex to avoid cutting off the content?

Thanks!

LLZN, 2019-05-14:
FIX Czech translation

barbeau, 2019-05-15:
@LLZN Thanks!  That fixed the "Accuracy" screen, and "Status" is better.  But, "Status" width still needs to be reduced by a few characters:

[image]

I know it's a challenge but just try to get something that still makes sense but fits.  If the abbreviation needs additional explanation please add this to the `help_text` string.

LLZN, 2019-05-15:
fix czech translation

barbeau, 2019-05-20:
Thanks @LLZN!  That's definitely better.  If you can shave another character off that would be ideal, but if not it's ok as-is too:

[image]

closed 2019-05-20 by barbeau

goldfndr, 2019-05-21:
@barbeau Unless there's a language in which the translation of "time" is longer than the translation of "accuracy" and other texts, or you anticipate other lengthy values, the space between columns 2 (Lat/Long values) and 3 (Time, abbreviation of TTFF) can be eliminated without fear of legibility impacts.

Or perhaps not -- I don't know if TableLayout permits varying column spacing. Perhaps setting the second column to shrinkable would work, as the last digit (or degree symbol) of the coordinates would rarely be used?

LLZN, 2019-05-21:
Hello, if I changed the phrase "Přesn. (H/V)" to "Přesnost", it would
shorten the characters.  But then the expression would be identical to
horizontal accuracy.  It is virtually impossible to reduce the number of
characters in the Czech language without losing any meaning.  Thank you L.

barbeau, 2019-05-21:
@LLZN That's fine, we can leave it as-is.

@goldfndr Good idea, I'll look at tweaking the layout.  It's been a while since I touched that, if ever...