barbeau/gpstest #296: Support entering Accuracy ground truth altitude in MSL

barbeau, 2019-06-05:
**Summary:** 

In the current Accuracy feature within GPSTest, you can enter a ground truth location for the app to compare against the GNSS calculated position to determine horizontal and vertical error.

Currently the altitude for the ground truth location must be entered in meters above the WGS84 ellipsoid, which is the way that GNSS receivers represent altitude.  WGS84 altitude is the only form of altitude officially supported by Android:
https://developer.android.com/reference/android/location/Location.html#getAltitude()

Unfortunately, most people are used to referencing altitude in the form of meters above mean sea level (MSL) - here's a good article describing the difference:
https://www.esri.com/news/arcuser/0703/geoid1of3.html

It would be more natural for many users to be able to enter ground truth altitude in MSL (see a user's post at https://groups.google.com/forum/#!topic/gpstest_android/gbSqAhoI8Xg), but unfortunately Android doesn't directly provide this data.

As a result, apps are left to try and determine MSL altitude on their own, frequently via:
1. Calculating it themselves - This is tricky because the difference between ellipsoid and MSL heights isn't constant - the difference varies depending where on Earth you are located.  So, you need a lookup table to find the difference based on your current location, and then you can use that value to determine the MSL altitude from the Android Location API.  See https://github.com/barbeau/earth-gravitational-model for a project I started working on to port some of the GeoTools library to Android to do this operation.  This implies bundling a approximately [700KB lookup table](https://github.com/barbeau/earth-gravitational-model/blob/master/src/main/resources/egm180.nor) within the app or setting up a server REST API to do the lookup remotely.
1. Parsing it from raw NMEA data - This is the approach that GPSTest takes. The [NMEA standard](https://www.gpsinformation.org/dale/nmea.htm) for GNSS receivers supports MSL altitude as part of `$GPGGA` sentences.  Android allows GNSS chipset OEMs to provide NMEA sentences via [an Android API](https://developer.android.com/reference/android/location/OnNmeaMessageListener). However, there is no guarantee that a particular device will support MSL altitude via NMEA - it's not required by Android.  The current behavior for GPSTest Status screen is that if the app gets a NMEA sentence update that contains MSL altitude it will display it, otherwise the field is hidden.

So, the design/implementation challenges here are:
1. Determine when to allow user to enter altitude in MSL - With our current implementation of MSL, the user's device must support altitude MSL via NMEA, but we don't know if the device supports it until we get NMEA updates within the app.  *Possible solutions:* We could look at dynamically changing the UI when we determine support, or look at changing the MSL implementation to calculate MSL within the app from WGS84 instead of relying on NMEA support.

A much simpler overall solution is if Android started directly supporting/requiring devices to provide altitude in MSL as part of the Location API.

**Steps to reproduce:** 

1. Select the "Accuracy" view in the navigation drawer

**Expected behavior:** 

Allow me to enter the ground truth altitude in meters above sea level

**Observed behavior:** 

The only supported input for altitude is meters above the WGS84 ellipsoid

**Device and Android version:** 

N/A

goldfndr, 2019-06-06:
Even worse, [some phones don't report MSL accurately](https://forum.xda-developers.com/showpost.php?p=79283870&postcount=9):

* [duplicates on Huawei P30 Pro](https://medium.com/@nignikool12/here-are-3-screenshots-from-huawei-p30-pro-f804682dbde6) (or [332,3 m](https://groups.google.com/forum/#!topic/gpstest_android/kVdGi6WyKlQ))
* [-55.1 m for both on Huawei Mate 20 Pro](https://medium.com/@lukspica/surprisingly-on-huawei-mate-20-pro-it-does-not-look-like-cf-is-visible-6b5d60d5d135) (or [147,5 m](https://medium.com/@gabrielgavril29/mate-20-pro-55118b550))
* [537,4 m on Huawei P20](https://medium.com/@ciaran.deignan/my-huawei-p20-got-an-upgrade-to-9-0-0-201-c432e10r1p16-still-nothing-in-the-cf-column-d7ee898dd70c)

(It is admittedly *possible* that ellipsoid and geoid altitude are identical at some of those coordinates.)

StefT3, 2019-06-07:
maybe a way:
http://epsg.io/8037

barbeau, 2019-06-11:
@StefT3 Thanks!  Looks like this is a hosted web service to do the conversion - I believe this is the format:
https://github.com/klokantech/epsg.io#api-for-trans

barbeau, 2020-04-30:
I've opened an issue on android-maps-utils to support general WGS84<->MSL altitude conversion - https://github.com/googlemaps/android-maps-utils/issues/704.

barbeau, 2023-02-13:
Support for WGS84->MSL altitude is live in Android U DP1: https://developer.android.com/reference/android/location/altitude/AltitudeConverter

We can also use this API to get the geoid offset by passing in a location with WGS84 altitude 0 - you will get the negative geoid offset back via Location.getMslAltitudeMeters().

barbeau, 2023-09-03:
Not stale

barbeau, 2024-03-22:
Not stale

barbeau, 2025-04-27:
Not stale