barbeau/gpstest #297: gps not turned off when 'closing' app

spaetz, 2019-06-05:
**Summary:**
While gpstest is running, the location icon is shown, ie the gps is turned on. When closing the app by pressing the home bitton or the 'app switch button', the icon is still shown and battery drains quicker. As soon as the app is 'swipe-closed' in the app overview or it is force-closed in t app details page, the location icon disappears.

**Expected behavior:** 
When gpstest is not in the foreground it should not keep the gps turned on.

**Observed behavior:** 
As long as the back is somehow active in the background, it keeps the gps turned on.

**Device and Android version:** 
Aquaris U plus (tenshi), LineageOS 14.1, no gapps.

Thanks for GPStest, it is really useful.

goldfndr, 2019-06-06:
Duplicate of #237; look in GPSTest's Settings for a "Stop GNSS in background" (or translated equivalent) near the bottom in the "GNSS" category.

(Also, see [my comment](https://github.com/barbeau/gpstest/issues/237#issuecomment-439209286) about how pressing Home isn't an Android method of closing an app, it puts the app in the background; the Back button can be used to close an app.)

barbeau, 2019-06-10:
Thanks for the feedback!  Yes, I'm closing this as a duplicate of https://github.com/barbeau/gpstest/issues/237, but I'll comment further there on a better implementation of background activity.

closed 2019-06-10 by barbeau