barbeau/gpstest #299: Run GPSTest as foreground Service when in background

barbeau, 2019-06-10:
**Summary:** 

GPSTest was originally implemented to remaining running in the background when you press the Home key.  As discussed in https://github.com/barbeau/gpstest/issues/237 and https://github.com/barbeau/gpstest/issues/297, this has been interpreted to be a feature, not a bug, as it can be helpful to prime GPS in some situations.

However, given the advances in Android since 1.5, background execution should be overhauled to the current Android best practices, which means running it as a foreground Service when in the background.  This will allow consistent behavior across devices and a clear notice to the user that GPSTest is still running in the background.

Some UI decisions need to be implemented here, including whether we prompt the user to ask if they want to continue running GPSTest in the background at some point, or just invert the default option and always have GPSTest stop running when the user puts the app in the background unless the user enables a setting to run in the background.

**Steps to reproduce:** 

Start GPSTest and hit the Home button

**Expected behavior:** 

Give me some notice that it's running in the background, and use an Android foreground Service

**Observed behavior:** 

App simply doesn't de-register the LocationListener, which on some devices means it continues running in the background, while on other devices it will be killed.

**Device and Android version:** 

N/A

barbeau, 2019-07-02:
As mentioned in https://github.com/barbeau/gpstest/issues/306, users would also like to be able to log to Android Studio and eventually files when the screen turns off - this is the same issue.

barbeau, 2019-09-17:
We also get to specify a notification when running in the background - a user requested something similar to the GPS Status app:

[image]

zemix, 2020-02-20:
Yeah, thanks for replay on Play Store, I personally absolutel support this topic, the way it is implemented in GPS Locker application for example! **_(duplicate my review)_**
_"Nice GPS testing app! Only one suggestion - add please ability to lock🔒GPS signal(status) after pressing back button (exit from app) for background staying Gps alive and don't loose 🛰 satellites! Of course as Option in Settings. It would be very useful, because for my own opinion as a solid user - your GPSTest app finds signal and lock satellites more faster and I am really love Sorting option in your app."_

barbeau, 2020-03-03:
Related to implementing ViewModel and Service:
* https://stackoverflow.com/questions/51486170/viewmodel-when-implementing-room-in-foreground-service
* https://github.com/android/location-samples/tree/master/LocationUpdatesForegroundService/app/src/main/java/com/google/android/gms/location/sample/locationupdatesforegroundservice

barbeau, 2021-04-15:
Given the most recent Android architecture principles and tools, it seems like the following is the best architecture:
1. Use a foreground service to listen for location updates
2. Send location and calculated metadata to a datastore that supports observables. Room with Flow (as of Room 2.1) seems like the best option (see [this](https://www.raywenderlich.com/9799571-kotlin-flow-for-android-getting-started) for an example).
3. Listen to the datastore within the Fragments using LiveData and ViewModels, which will then populate the views when new updates get send to the datastore.

This design is discussed here, with the below demo app using `LocalBroadcastManager` instead of Room with Flow:
https://github.com/googlecodelabs/while-in-use-location/issues/12

[DataStore](https://developer.android.com/topic/libraries/architecture/datastore) is a possible alternative to Room, although with some of the more complex metadata it seems like Room is a better option.

barbeau, 2021-04-30:
Good example of LifecycleService:
https://github.com/gs-ts/TrackMyPath/blob/c3bd825d4c3f2eaea37ca5e659eb6127f5778e14/app/src/main/java/com/gts/trackmypath/presentation/service/LocationService.kt

Some other good resources:
* Room with Flow - https://www.raywenderlich.com/9799571-kotlin-flow-for-android-getting-started
* DI using Hilt - https://developer.android.com/training/dependency-injection/hilt-android

barbeau, 2021-05-18:
I've created a demo here (based on Google's code sample) that uses Room and Flow for location updates:
https://github.com/barbeau/while-in-use-location

barbeau, 2021-06-28:
Here's a good example showing StateFlow with loading, etc. states:
https://github.com/philipplackner/StateFlow

...from this YouTube video:
https://www.youtube.com/watch?v=Qk2mIpE_riY&t=325s

There is also another example I need to find that explained how to use MutableStateFlow without a backing private variable, IIRC. Maybe it was an Android dev team Kotlin video?

barbeau, 2021-07-02:
Good example of managing service restarts using KTX:
https://github.com/android/health-samples/blob/f32c8956daa3f9e2c9bce6cf45102a8ae34bc22e/ExerciseSample/app/src/main/java/com/example/exercise/ExerciseService.kt

barbeau, 2021-07-03:
Another possibility is to wrap the location listener in its own LocationRepository (e.g. instead of using Room).

Some guidance in this direction:
* https://youtu.be/OmHePYcHbyQ  - How to handle get most recent location
* https://developer.android.com/codelabs/building-kotlin-extensions-library#0 - Code lab for above video. Includes request location updates.
* https://github.com/googlecodelabs/kotlin-coroutines/blob/master/ktx-library-codelab/step-06/myktxlibrary/src/main/java/com/example/android/myktxlibrary/LocationUtils.kt
* App using flow from location provider - https://github.com/googlecodelabs/kotlin-coroutines/blob/master/ktx-library-codelab/step-06/app/src/main/java/com/example/android/kotlinktxworkshop/MainActivity.kt
* Towards end of top video there is a location repo with location flow KTX is shown, but doesn't seem to be link to code? At 9:18

barbeau, 2021-07-15:
A good video visually explaining the differences between the different types of Flows, in context of updating locations on a map:
https://youtu.be/RoGAb0iWljg

And the corresponding repos:
* https://github.com/Ryan-Pierce/MapsSharedFlowDemo
* https://github.com/Ryan-Pierce/MapsStateFlowDemo
* https://github.com/Ryan-Pierce/FlowAndLiveDataSimpleDemo

barbeau, 2021-11-10:
Here's an article I wrote on wrapping the location listener in its own LocationRepository with Kotlin callbackFlow:
https://barbeau.medium.com/kotlin-callbackflow-a-lightweight-architecture-for-location-aware-android-apps-f97b5e66aaa2

Link to example on GitHub is in the article. I'm using this pattern in GPSTest.

closed 2021-11-11 by barbeau