barbeau/gpstest #306: GPStest is instantly stopped on screen off

haroon-ali, 2019-07-02:
**Summary:** 

I try to make use of GPStest logging feature. But once the screen is turned-off logging is instantly stopped.
GPStest should be kept running unless I explicitly stop it.

**Steps to reproduce:** 

Run GPStest, watch logcat, turn screen off, no more logging.

**Expected behavior:** 

Keep running and logging.

**Observed behavior:** 

It stops working.

**Device and Android version:** 

Xiaomi Redmi Go,
Oreo 8.1 Go edition
Stock rom V10.2.7.0
Google Play Services 14.5.74

**Screenshots:** 

N/A

haroon-ali, 2019-07-02:
Follow-up:
I could work-around the reported problem by the following steps:
1. Rooted the phone using magisk
2. Modified GPStest manifest file and added android:persistent="true" there, and rebuilt the apk.
3. Installed my modified apk, and using app-systemizer magisk module and made it a system app

Now it's not killed on screen off :+1:

haroon-ali, 2019-07-02:
Follow-up 2:
It seems that the previous solution is not always working :(

barbeau, 2019-07-02:
@haroon-ali Yes, currently behavior will be a bit erratic when the app is minimized (including when the screen is turned off), as it really wasn't initially designed to run in the background.  I have an open issue here for true background execution support - https://github.com/barbeau/gpstest/issues/299. I added a comment there for your use case, so I'm going to close this issue.

For now the only workaround is to keep the screen on (see the option in the settings).

closed 2019-07-02 by barbeau