barbeau/gpstest #307: Allow smaller than a second GPS fix intervals

haroon-ali, 2019-07-02:
**Summary:** 

GPS fix interval  values smaller than 1 second, like 0.2 for example, doesn't work.

**Steps to reproduce:** 

In settings, set GPS fix interval to 0.2. Go to main screen, it says "Starting with GPS interval = 0.2" But actually it only fixes and logs once per second.

**Expected behavior:** 

To get fix according to interval setting.

**Observed behavior:** 

It gets a new fix only once per second. This can be clearly seen on logs with timestamps, where it updates only once per ~1000 microsecond.

**Device and Android version:** 

Test on multiple different devices. One of them is Redmi Go, Stock ROM, Oreo 8.1 Go.

brucechase, 2019-07-02:
I have built numerous test programs that incorporate about every
conceivable function calls having to do with GNSS and GPS for the Android
phone.The tests programs were using the latest Android version (P) at the
time.  The results were always the same.  No test program could gain
accuracy greater than 1 Hz (1 per second).

Like you, I would like 5 Hz (5 times per second) or better for a specific
application.Whereas the newer multi-frequency phones allow greater accuracy
with multiple satellites, there appears to be no greater Hz rating. Any
greater Hz rating would become approximations.

In looking through some of the Android source code, I can find no example
of greater than 1 Hz. If you find it, please let me know.

barbeau, 2019-07-02:
@haroon-ali thanks for the report!  As @brucechase mentioned, unfortunately this is a property of the device you're testing on.  The Android platform and GPSTest both support request a sub-second interval, but ultimately it's the device that controls how frequently an updated position is sent to the app. So you'd need to ask your device OEM why it doesn't support sub-second update intervals.  If you could add the other device models you've tested on it may help others who are wondering if their device supports this.

There is a related discussion on the Google Group here:
https://groups.google.com/d/msg/gpstest_android/OBUjBnbJejw/551AFDAmAgAJ

closed 2019-07-02 by barbeau