barbeau/gpstest #308: Feature request: Support sending nmea messages to external devices over usb/serial

Jaaaky, 2019-07-04:
**Summary:** 

I'd like to get nmea messages - and other types of logs supported by GPStest - over USB serial connection - for example FTDI connected to USB - to be able to analyze through external devices.

**Steps to reproduce:** 

Connect FTDI cable on android OTG USB

**Expected behavior:** 

GPStest handles the device and sends logs over it.

barbeau, 2019-07-08:
@Jaaaky Thanks for the suggestion!  This probably wouldn't be too hard to implement - here's a library that looks promising:
https://github.com/felHR85/UsbSerial

I don't have the equipment to test though - if I work on this, would you be able to test on real equipment and provide feedback on whether it works?

haroon-ali, 2019-07-08:
Yes, I can. I can send you some list of usb vendor:device numbers.

Jaaaky, 2019-07-08:
I have the equipment, yes. So I can test.

Jaaaky, 2019-07-08:
@barbeau They work with the mentioned library.

lindroidux, 2019-10-13:
@barbeau Could you pls include this feature over Bluetooth/network?

barbeau, 2019-10-15:
@lindroidux I've started this here:
https://github.com/barbeau/gpstest/tree/serial

...but hit an issue with the app crashing, and I can't tell if the code's not right or it's just because I don't have a device connected. So it's going to be hard to implement this without a physical device to test on myself.

Jon8RFC, 2020-01-13:
@barbeau Would you like one of these?
https://smile.amazon.com/GlobalSat-BU-353-S4-USB-Receiver-Black/dp/B008200LHW/ref=sr_1_1?keywords=globalsat+usb+5hz&qid=1578885807&sr=8-1

You could add it to a new amazon wishlist and post it here.  I'm not sure where you live, but if international shipping is added on Amazon, that might be out of my budget.

I'm interested in getting complete data, viewed in your app, from usb AND bluetooth receivers.  I just want to figure out which one works best for me to use as a "mock location" device in my car.

Thanks for your work on this application!

goldfndr, 2020-01-18:
https://smile.amazon.com/dp/B07P8H27CF/ looks like a slightly better deal, but I don't know how well it'd compare to the one you mentioned -- it looks like it'd need [an outProtoMask with outNmea set](https://www.u-blox.com/sites/default/files/products/documents/u-blox8-M8_ReceiverDescrProtSpec_%28UBX-13003221%29_Public.pdf) to output NMEA. It has a customer answer of not working with Android, but so does the BU-353-S4.

Jaaaky, 2021-10-30:
> Closing this. Please reopen if you believe it should be addressed. Thank you for your contribution.

Please re-open this.. I see no reason for close.. It was already scheduled for 3.9 but simply you decided to close it.

Bluetooth/Serial sharing of GPS NMEA with background service would be so great so one can do further "Live" studies ..

gdt, 2022-11-07:
Hey bot: Despite no activity, this is still valid.

barbeau, 2023-12-19:
Not stale

barbeau, 2025-01-24:
Not stale