barbeau/gpstest #310: Screen rotation turns of satellite updates on Status screen.

rick-gse, 2019-07-10:
**Summary:** 

Using version 3.3.2 of GPSTest

Rotating screen turns off the satellites updates on the Status screen.

**Steps to reproduce:** 

Show status screen in portrait mode, satellites are updating.
Rotate screen to landscape. Satellites stop updating.

**Expected behavior:** 

Satellites should stay updating after rotation.

**Observed behavior:** 

Satellites stop updating.

**Device and Android version:** 

Google Pixel 3XL.
Android 9

barbeau, 2019-07-10:
@rick-gse Could you include a screenshot of your Settings in GPSTest? And if you uninstall and re-install can you reproduce this every time with the default settings?

I can't reproduce this in v3.3.3 on a Samsung Galaxy S8+ w/ Android 9.  You might want to try v3.3.3 as well (rolling out to production now, but you can get it immediately if you join the beta group at https://play.google.com/apps/testing/com.android.gpstest). It has minor changes though that shouldn't impact this.

barbeau, 2019-07-10:
Note that the resetting of satellites on Status is expected - the GPS re-initializes when you rotate the app.  So if you're in a place where it's hard to get a fix, you may need to wait a while for the satellites to re-appear.

rick-gse, 2019-07-10:
Re-installing fixed it, so I did some testing, it's the "Stop GNSS in background" setting. When this is ON, rotation stops the updates, when it is off, rotation does not stop the updates.

[image]

[image]

[image]

rick-gse, 2019-07-10:
Also ran it in the android emulator using the current code, and it did the same thing.

rick-gse, 2019-07-10:
This seemed to fix it, change onPause in GpsTestActivity.java to add a check for orientation change:

        // Check if the user has chosen to stop GNSS whenever app is in background
        if (**!this.isChangingConfigurations() &&** Application.getPrefs().getBoolean(getString(R.string.pref_key_stop_gnss_in_background), false)) {
            gpsStop();
        }

barbeau, 2019-07-11:
@rick-gse Thanks!  I'll take a look.

barbeau, 2019-07-11:
@rick-gse Thanks for the proposed code!  I looked at a few possible solutions and given the complicated history of how GPSTest currently "works" in the background this is the simplest solution.  Ultimately background execution will be properly fixed in https://github.com/barbeau/gpstest/issues/299.

closed 2019-07-11 by barbeau

gse-rick, 2019-07-11:
Glad to help. I also looked at also not stopping the listeners, but was worried about unintended consequences.

commit 4633d352 (2019-07-11): Fix #310 - Only stop GPS if user set preference AND didn't rotate device