barbeau/gpstest #313: Feature request: list GNSS which my phone supports

Mikaela, 2019-07-15:
**Summary:** 

I wish the Status or a similar tab told me which GNSS my phone supports directly saying yes/no.

E.g.:

* GPS: yes
* Glonass: no
* Galileo: no
* etc.

**Steps to reproduce:** 

Open GPS Test for the first time without reading much about GNSS with question whether your phone supports Galileo or not. Even better test may be giving your phone with GPSTest open and asking them to quickly tell you what navigation systems the phone support.

**Expected behavior:** 

Anyone is able to quickly/easily tell what GNSS the phone supports.

**Observed behavior:** 

I see my location & technical information I don't entirely understand. I see ID, GNSS/flags etc.

As I only see the US flags (and satelite IDs 7 8 10 13 15 16 20 21 26 27 29 30), after thinking about this and reading for some time I understand that my phone only supports GPS (and not Galileo which I understand to be having service disruption, but due to comparing with someone else online I know that the satelites appear within the app regardless). Due to my phone being low end device marketed to developing countries I am not particularly surprised.

**Device and Android version:** 

* Nokia 1 (TA-1047) 
* Android 9 Pie Go Edition (2019-06-05 security patches)
* stock ROM
* Google Play Services version 18.3.81 (110304-257258062)

barbeau, 2019-07-19:
@Mikaela Thanks, this is a good idea.  It could also cover displaying the info discussed inhttps://github.com/barbeau/gpstest/issues/265 and https://github.com/barbeau/gpstest/issues/295.

The caveat here is that Android APIs only tells applications about satellite properties in view of your device right *now* - there isn't any way to query general support for satellites.  So, for example, if your device hardware and software supports QZSS, but you're not in a location where QZSS satellites are visible, GPSTest has no way of knowing that your device supports QZSS.

But a simplified view of what the device supports based on satellites currently in view is a good idea.

barbeau, 2019-10-23:
I've started working on logic that can support this at https://github.com/barbeau/gpstest/pull/355.

barbeau, 2021-06-01:
So here's a very rough UI mockup for this screen showing device GNSS support. It would appear as another top-level navigation item, likely "Device Info", along side "Status", "Map, "Sky".

Any feedback/suggestions/alternate proposals welcome!

[image]

barbeau, 2021-07-06:
Interesting idea proposed here:
https://groups.google.com/g/gpstest_android/c/W1OB-L_o5kQ/m/LKO80NPAAQAJ

Could we score overall device GNSS capabilities?

>adding a numeral field with a formula to score  total gps capabilities (for example bands, frequencies,  antennas,... )

Not sure what that exact formula would look like, but ideas are welcome.

Mikaela, 2022-06-02:
I would still find this feature helpful

barbeau, 2023-05-18:
Android U adds new methods to the [`GnssCapabilities`](https://developer.android.com/reference/android/location/GnssCapabilities) API that will allow us to read things like signal support directly from the API versus relying on real-time observation of signals, so we should use that as well.

barbeau, 2023-12-05:
Not stale

Mikaela, 2024-06-22:
I do still sometimes find myself missing this only to find out there is an open issue.

barbeau, 2024-06-23:
Agreed, not stale