barbeau/gpstest #314: Support more granular deletion of assist data

barbeau, 2019-07-28:
**Summary:** 

Looking at the Android source code, it appears that more granular options for deleting GNSS assist data are now available on newer versions of Android (I'm not sure which as of right now).

Previously, an app could send an "extra" command of `delete_aiding_data` to the `LocationManager` and on devices that supported this command the device would delete the assist data (an aside - I'm not sure why some devices don't support this command).

Newer versions of Android source code support additional granularity of deletion of specific types of assist data. Here's the link:
https://android.googlesource.com/platform/frameworks/base.git/+/master/services/core/java/com/android/server/location/GnssLocationProvider.java

...and the code itself:

~~~
    private boolean deleteAidingData(Bundle extras) {
        int flags;
        if (extras == null) {
            flags = GPS_DELETE_ALL;
        } else {
            flags = 0;
            if (extras.getBoolean("ephemeris")) flags |= GPS_DELETE_EPHEMERIS;
            if (extras.getBoolean("almanac")) flags |= GPS_DELETE_ALMANAC;
            if (extras.getBoolean("position")) flags |= GPS_DELETE_POSITION;
            if (extras.getBoolean("time")) flags |= GPS_DELETE_TIME;
            if (extras.getBoolean("iono")) flags |= GPS_DELETE_IONO;
            if (extras.getBoolean("utc")) flags |= GPS_DELETE_UTC;
            if (extras.getBoolean("health")) flags |= GPS_DELETE_HEALTH;
            if (extras.getBoolean("svdir")) flags |= GPS_DELETE_SVDIR;
            if (extras.getBoolean("svsteer")) flags |= GPS_DELETE_SVSTEER;
            if (extras.getBoolean("sadata")) flags |= GPS_DELETE_SADATA;
            if (extras.getBoolean("rti")) flags |= GPS_DELETE_RTI;
            if (extras.getBoolean("celldb-info")) flags |= GPS_DELETE_CELLDB_INFO;
            if (extras.getBoolean("all")) flags |= GPS_DELETE_ALL;
        }
        if (flags != 0) {
            native_delete_aiding_data(flags);
            return true;
        }
        return false;
    }
~~~

So if we send the corresponding key/value pairs in a bundle with the `delete_aiding_data` command, on supported devices it should delete only that type of assist data. Right now the extras bundle is `null` so the current command in GPSTest deletes all assist data.  I'm assuming that supporting additional granularity in GPSTest won't change anything on devices that don't support this command.

**Steps to reproduce:** 

From the nav drawer select "Delete assist data"

**Expected behavior:** 

Give me an option to choose which type of assist data to delete

**Observed behavior:** 

A command to delete all assist data is always sent to Android

**Device and Android version:** 

N/A

barbeau, 2019-07-28:
Looks like this goes back to at least Android 8.0:
https://github.com/aosp-mirror/platform_frameworks_base/commit/282b594b688c731411622b252abbcb616901ae44#diff-f5c0ecdf63cc4f7857daa6525648c38b

Here's Git history from the GitHub mirror:
https://github.com/aosp-mirror/platform_frameworks_base/commits/master?after=8aa2b1dffd2eac9749116e92f5f9d6a7a62c5365+34&path%5B%5D=services&path%5B%5D=core&path%5B%5D=java&path%5B%5D=com&path%5B%5D=android&path%5B%5D=server&path%5B%5D=location&path%5B%5D=GnssLocationProvider.java

sm4rk0, 2019-07-29:
And here's the file history before it was renamed in December 2015 (showing that `deleteAidingData` didn't change at least the last 11 years):
https://github.com/aosp-mirror/platform_frameworks_base/blame/efea96719708d3e1874ca372d574089cf84f2f57/services/core/java/com/android/server/location/GpsLocationProvider.java#L1242

barbeau, 2019-07-30:
@sm4rk0 thanks! You're right, looks like this actually goes way back - I'm guessing I just never looked at this part of the AOSP code before.

barbeau, 2024-06-22:
Closing this because I don't have an actual use case.

closed 2024-06-22 by barbeau