barbeau/gpstest #316: OSMDroid - No map tiles on Accuracy view

barbeau, 2019-07-31:
**Summary:** 

If you open the OSMDroid build of GPSTest and go to the "Accuracy" view, the map tiles are blank.

I'm currently using Mapbox's satellite tiles for this, and apparently I've hit the free tier quota and they are no longer serving tiles. I've asked Mapbox about providing free credits but haven't heard back.

Another option might be OpenMapTiles:
https://openmaptiles.com/

**Steps to reproduce:** 

In OSMDroid release, go to Accuracy view

**Expected behavior:** 

See satellite map tiles

**Observed behavior:** 

Map tiles are blank

**Device and Android version:** 

N/A

barbeau, 2019-08-14:
Re-opening this - apparently the F-Droid version is popular, we've now exhausted our free tier of the MapTiler account:

>Your MapTiler Cloud account has used 100% of the FREE plan monthly request limit (100 000) in this billing period.

I may just revert the Accuracy view to the free OSM map tiles.

mgw2013, 2020-02-15:
Issue still open.

cewbdex, 2021-05-03:
Are there any advantages of MapTiler over OSM based services?

barbeau, 2021-06-22:
@cewbdex Sorry for the delayed response, somehow I missed your comment. IIRC MapTiler is an OSM-based service - it serves tiles based on OSM data. Is there an alternate service you'd suggest?

Ideally it would be free for open-source projects, as this app is free.

barbeau, 2024-12-04:
@pipatron Thanks for the info! Yes, I think this has changed since the last time I looked at it. I'll give tile.openstreetmap.org a shot.

pipatron, 2024-12-04:
> If usage is/was exceeding MapBox's quota and MapTiler's quota then using OSM's tile server doesn't seem sustainable. It is **not** the go to source if usage is high. But it might not be a bad idea if it's a user-selected choice among tile servers.

Why do you think that? MapBox and MapTiler are completely independent servers and organizations with their own agenda and they didn't set these limits because they lacked bandwidth. It's not like they are caching servers for the OSM tile server. OSM tile servers serve the whole world's OpenStreetMap and are designed for real-time map distribution. GPSTest's usage would be an unnoticeable drop in the ocean when it comes to bandwidth.

badbettty, 2026-02-10:
thank you for all the work going/put into this app.

just for info and of no further use: using latest FDroid version (on moto g34 5g)  and this issue is still occuring.

commit 518cdea6 (2019-08-01): Fix #316 - Switch from MapBox to Maptiler for Accuracy satellite tiles