barbeau/gpstest #317: Status view - On narrow screens rows spill over

barbeau, 2019-07-31:
**Summary:** 

On very narrow devices such as the Lenovo Z6 Youth, on the Status screen the rows barely spill over:

[image]

**Steps to reproduce:** 

Open app to Status screen on narrow screen device such as Lenovo Z6 Youth

**Expected behavior:** 

All info for one satellite should appear on one row

**Observed behavior:** 

The degrees symbol barely spills over to the next line.

**Device and Android version:** 

Lenovo Z6 Youth

barbeau, 2019-07-31:
Well, this might be a font issue again - I created an emulator with 6.30-inch display with a resolution of 1080x2340 pixels (from https://gadgets.ndtv.com/lenovo-z6-youth-edition-price-in-india-13079) and it looks like this:

[image]

barbeau, 2019-08-01:
So I can come close to reproducing this with a 1018 x 2340 xxhdpi setting, although the emulator only fills the columns with "30.0" values, and I need 3 digits like "130" to push the character to the next line.

I'm going to fix this by clipping out the ".0" for azimuth and elevation, which also addresses a previous request for this feature.

closed 2019-08-01 by barbeau

goldfndr, 2019-08-02:
Thanks. For the record, that was #130 (my comments). Not ideal (I also am prone to bugging the other direction; see https://github.com/wiglenet/wigle-wifi-wardriving issue 27) but certainly good enough for every screenshot I've seen.

barbeau, 2019-08-03:
Just to clarify what the implementation actually does - the number is rounded to a single digit the same as before, but then the text is altered so if it ends with `.0` that part is removed.

So the value `13.0` will appear as:
* `13`

...but the value `13.1` will still appear as:
* `13.1`

commit 5504c41f (2019-08-01): Fix #317 - Remove ".0" from elevation and azimuth in Status view