barbeau/gpstest #318: Accuracy - Add fine-grained control over ground truth location marker

barbeau, 2019-07-31:
**Summary:** 

From a user:

>Playing with setting "known" positions with GPStest, I keep wishing I could nudge the touch-selected location, when I have dropped the map pin a little adrift from where I intended. It would be nice to have up/down arrows next to the edit fields to increment/decrement the last digits displayed with taps for single steps or long presses for longer jumps. I'm not sure if I'd be more accurately testing my device, or more finely resolving the satellite image locational discrepancy, but it would be good play value either way.

**Steps to reproduce:** 

1. Go to Accuracy view
1. Tap on the map to set a ground truth location

**Expected behavior:** 

Give me a more precise way to adjust the map marker location

**Observed behavior:** 

To adjust the map marker location you have to tap far away from the current marker location, and then try tapping in the first location again.

**Device and Android version:** 

N/A

AdiityaAndre, 2020-07-26:
I've had some issues. Either it is related with this.
Ground truth location can not be edited while we already had pin on location. The field is greyout. We can only tap on the maps  when we want to edit the ground truth location.

barbeau, 2020-08-07:
@AdiityaAndre Sorry about this - it was a bug in an Android library. This is fixed in the latest v3.7.3 release which is now in beta, if you want to test it out!

AdiityaAndre, 2020-08-08:
@barbeau Thanks sir..

closed 2022-10-28 by stale[bot]