barbeau/gpstest #320: Room for strings is too limited, after transaltion data doesn't fit

Meteor0id, 2019-08-07:
I have attempted to keep the Dutch translations short, but sometimes a text string just needs some length in order to be informative. Is there any possibility of using the available space in a more tolerant matter?

[image]
[image]

barbeau, 2019-08-07:
This is effectively a duplicate of https://github.com/barbeau/gpstest/issues/176, so let's talk more about configurable Status displays there.

In short, for the current Status view just abbreviate as much as you need to to get the info all on the screen, and explain the abbreviations in the Help text.

closed 2019-08-07 by barbeau

paulle, 2019-08-08:
Making linebreaks and leaving the selection of the font size to the user would be a good solution.

goldfndr, 2019-08-14:
If "E" in the accuracy labels is for Estimated then I suggest dropping it.

* It's relatively redundant — all of the accuracy values are estimates.
* It's not consistent — the speed and bearing accuracy labels don't include it.
* It can create confusion — the Help/About descriptions don't include it, so there's a mismatch.

Meteor0id, 2019-08-14:
No, the E does mot stand for estimated. Please go to help > about to find tge well hidden legend which explains the way too short abbriviations.

barbeau, 2019-08-14:
@goldfndr "E" does stand for "estimated" in the context of horizontal and vertical error.  I added this when I added the "Accuracy" feature to GPSTest to try and differentiate between the actual and estimated accuracy values.  In other words, I didn't want someone to set a ground truth value in Accuracy and then come back to Status and think that real accuracy was being displayed in Status.  Confusion between estimated and actual accuracy is a huge problem and this was my small attempt in a limited timeframe to address it.  I'd welcome better ideas for this in https://github.com/barbeau/gpstest/issues/176.

And you're right, this isn't in the Help/About section, so I need to update that.