barbeau/gpstest #322: Letters for Flags not translatable

Meteor0id, 2019-08-07:
In an attempt to make GPSTest more accessible to users in foreign languages, it might help if the letters refer to words in their own language instead of English, after all the rest of the interface is also translated. Currently the letters for flags: A E and U, are not available for translation in Transifex.

barbeau, 2019-08-09:
@Meteor0id thanks for the feedback!  

What do other translators think about this?  Would it be better to translate A, E, and U flags (for "almanac", "ephemeris", and "used) for each satellite to the local language?

goldfndr, 2019-08-10:
A more radical option is to use emoji. For Android 4.3 compatibility, I suggest:

* [💖 :sparkling_heart:](https://emojipedia.org/google/android-4.3/sparkling-heart/) (sparkling heart) Almanac (general health and orbit)
* [📡 :sattelite:](https://emojipedia.org/google/android-4.3/satellite-antenna/) (sattelite antenna) Ephemeris (position within orbit)
* [🔒 :lock:](https://emojipedia.org/google/android-4.3/lock/) (locked) Used

[atom](https://emojipedia.org/atom-symbol/) would be nice for symbolizing orbits but it was introduced in Android 6.0.

barbeau, 2019-08-13:
@goldfndr Yes, ideally we'd use a visual representation instead of text.  I'd welcome any general ideas for images too (not just emojis) or a different graphical representation, although they would need to be small to fit.

Meteor0id, 2019-08-14:
I would normally encurage using simple icons, however representing emprisis data, usage and almanac data in icons sounds impossible to me. If the icons won't make sense, we shouldn't use icons, because they could lead to confusion or misinterpretation.

If someone manages to design icons which make sense, ok. Otherwiae lets stick with letters which represent the word.

closed 2022-10-28 by stale[bot]