barbeau/gpstest #324: Send Feedback menu makes no sense, Feedback goes to GitHub, there is already a button for that

Meteor0id, 2019-08-07:
The title of this issue said it all. The menu could be renamed "share this app" or just be removed.

barbeau, 2019-08-07:
@Meteor0id This might be a translation issue - the bottom 4 options in the navigation drawer should be (in order top to bottom):
* Visit us on GitHub
* Settings
* Help
* Send Feedback

Are you talking about the bottom option?

In the English version when I tap on the bottom option it opens an email Intent to a feedback email address.

Meteor0id, 2019-08-07:
I get a general share intent on Send Feedback. My email app is not opened directly. Seems the issue is how the intent is handled.

I am on Android 9, using an app called K9 for email.

barbeau, 2019-08-07:
It's an ACTION_SEND Intent with Intent.EXTRA_EMAIL extra:
https://github.com/barbeau/gpstest/blob/7dd6b9ccdf7e4e3c4483064d8a0e226b68919fe3/GPSTest/src/main/java/com/android/gpstest/util/UIUtils.java#L282

I can see devices handling this Intent differently, but it definitely shouldn't direct to Github. 

You're sure the bottom option opens to Github if you choose an app from the picker?

Meteor0id, 2019-08-08:
Currently thr feedback button is confusing. It opens a menu to choose an app or contact and than fais with an error. It should open the email app, but it doesn't.

Maybe tjis happense because the wrong intend action is being used. Should it not be Sendto instead of send?

barbeau, 2019-08-08:
Yes, looking at:
https://developer.android.com/guide/components/intents-common#Email

...ACTION_SENDTO would work here, with:

`intent.setData(Uri.parse("mailto:")); // only email apps should handle this`

I originally had SEND because I was planning on inclduing metadata in an attachment, but then decided not to, so this would work.

Meteor0id, 2019-08-08:
The problem is Intent.createChooser
Most Android devs advertise this as the best thing to do because "the user should decide waht app he wants to use". The issue is that his allow the user to pick whatsapp or Signal or any meesaging app as well as his e-mail app.

Since feedback can only be send using e-mail, lets remove Intent.createChooser .

commit 4a57926d (2019-12-15): Fix #324 - Limit feedback to email apps (#360)
closed 2019-12-15 by barbeau