barbeau/gpstest #325: Enable/Disable button same color as background

Meteor0id, 2019-08-07:
Should be a different color, as it is very hard to discover the switch this way

Sorry for opening issue tickets about little things like these, I don't have time to write a PR right away and by opeing an issue we can atleast keep track of the issue. I might write the PR later.

barbeau, 2019-08-07:
IIRC it's linked to the theme coloring, and I had to make a compromise between this and other UI components.  PRs are welcome for improvement, although you'll need to review the whole app when changes are made to make sure it's not making something else harder to see.

Could you add a screenshot so others know what you're talking about?

Meteor0id, 2019-08-07:
[image]

Meteor0id, 2019-08-07:
[image]

commit 9689ee9b (2021-09-10): fix: Use SwitchMaterial instead of Switch for better styling

Closes #325
closed 2021-09-10 by barbeau