barbeau/gpstest #327: Dutch translation for Status pushes text offscreen

barbeau, 2019-08-13:
**Summary:** 

This is a continuation of https://github.com/barbeau/gpstest/issues/320.

@Meteor0id I believe your translation changes pushed the text offscreen in the Status view in the latest Dutch translation updates:

[image]

Is this something you want to fix in the translation to abbreviate these fields again, or should I roll back to a previous Dutch translation that fit on the screen?

**Steps to reproduce:** 

1. Start app in Status view on Dutch locale (or select Dutch via Settings->Language)
1. Wait for GNSS fix

**Expected behavior:** 

Text for location in top card should fit on screen

**Observed behavior:** 

Most of the 2nd column is pushed off screen

**Device and Android version:** 

Samsung Galaxy S8+ with Android 9

Meteor0id, 2019-08-14:
Yes I noticed this and I already opened a ticket for it.
I coupd shorten the abbriviations, but that would make them completely unreadable. Understanding a couple letters would be guesswork. At that point we might as well assign every piece of data a number so you can look up the number in the legend. (The legend which is well hidden in help > about > faq)

Dutch abbriviations are inhernetly longer than English abbriviations. I strongly urge to introduce linebreaks.

Previously you mentioned  "information density" . Language can not be abbriviated indefinetly, if the text becomes too short, information goes lost. That doesn't make it more dense information, that makes it less information.
Linebreaks woupd maybe cause the status card to increase in height by 3 lines. If arranged intelligently, and if margins and paddings are optimalized, the impact on the layout can be reduced even further. I don't think line-breaks are a big problem at all. I would recommand adding line breaks. This applies to the status card and the bar on the bottom of the accuracy view.

barbeau, 2019-08-14:
@Meteor0id Proposed re-formatting of Status prioritized for information density is welcome if you'd like to do this as part of https://github.com/barbeau/gpstest/issues/176.

In the mean-time, though, we need to restore this to readable status for current users.  

Please let me know by end of the day on Friday of this week if you want to update the Dutch translation to bring the data back on-screen or if you want me to roll back to the previous Dutch translation that fit on the screen.  Thanks!

Meteor0id, 2019-08-15:
> In the mean-time, though, we need to restore this to readable status for current users.
>
>Please let me know by end of the day on Friday of this week if you want to update the Dutch translation

I have adjusted the translations. I haven't tested it, but I think it will fit. If a test build reveals it doesn't I'll shave off a few more letters.

barbeau, 2019-08-15:
@Meteor0id Great, thanks!  It's close, but needs a few more letters off:

[image]

Meteor0id, 2019-08-15:
It hurts my feelings to mangle my language even further. I tried to make them even shorter. Can you please also reduce the whitespace between the label and the value?

barbeau, 2019-08-16:
@Meteor0id Just to make you happy, I'll shave the whitespace down between the label and value :).

Thanks for your additional edits too - between these both, it fits! (at least on a Samsung Galaxy S8+). Thanks for fixing this!

[image]

commit 55b66709 (2019-08-16): Fix #327 - Update shortened Dutch translation, other translations
closed 2019-08-16 by barbeau