barbeau/gpstest #328: Set field width limits in Transifex

barbeau, 2019-08-13:
**Summary:** 

In issues like https://github.com/barbeau/gpstest/issues/327, some translations that get wordy push text offscreen.  For fields that have a constrained width, we should set those widths in Transifex.

Meteor0id, 2019-08-14:
I don't think constraints are nessesary. Translaters and reviewers know their space is limited. The reason we see things get pushed of screen is not poor translations, the issue is that the layout isn't flexible enough to handle lenght differences wich are simply inhearent to localization.

paulle, 2019-08-15:
Why not putting linebreaks in GPSTEST there text or output is shown. If text gets longer the user only have eventually to scroll a little bit more. I believe there are some languages there text gets longer than in English. For sure I know this for German.
Even for using GPSTEST with bigger fonts it would be an advantage (text woudn't be "cutted").

closed 2022-04-11 by stale[bot]