barbeau/gpstest #331: Add compass degrees to Sky view

barbeau, 2019-08-15:
**Summary:** 

Request from user:

>The only flaw is that it lacks the indication of degrees in the compass, like other similar app. 

**Steps to reproduce:** 

Look at Sky view

**Expected behavior:** 

Draw numbers on the compass (exact amount and design TBD) for the headings

**Observed behavior:** 

Only a north arrow is shown

**Device and Android version:** 

N/A

paulle, 2020-02-25:
[image]

paulle, 2020-02-25:
I did this as a svg. Could it be a possible replacement for the current windrose in skyview?

paulle, 2020-02-25:
x-special/nautilus-clipboard
copy
file:///home/chris/Downloads/compassrose.svg

barbeau, 2020-02-26:
@paulle Thanks for putting that together! Possibly - I need to revisit this code. Currently it's just drawing on a raw Canvas, and I'm not sure off the top of my head how easy it would be to layer a vector graphic on top.

Could you put the SVG XML in a GitHub Gist? Or share a link to it via Google Drive or something similar?

paulle, 2020-02-27:
https://drive.google.com/file/d/17YLEzUqp9zn5z7XV-b3BVN_xQPHmPqOK/view?usp=drivesdk

paulle, 2020-02-27:
I had some problems with my google drive. For this reason I sent you an e-mail wit the files attached.
barbeaudev@gmail.com

paulle, 2020-02-28:
[image]

closed 2021-10-02 by stale[bot]

cjc7373, 2023-04-24:
This feature is nice to have. At least show N, E, W, S letters instead of just an arrow..