barbeau/gpstest #334: Terminology: Bearing, course or azimith

Meteor0id, 2019-08-16:
I noticed the status card speaks of bearing. But is this actually correct chose of terminology?

Allow me to share what I think I know:
- Bearing is and angle between a point of reference and a point being described. It can be expressed as a number ranging from 0 to 360 or as an offset to a cardinal direction (South 64 degrees East). A Bearing describes a direction between A and B, in which A is usually your current location and B the destination. Bearing assumes a direct line. Bearing does not describe movement, it describes an intent.

- Azimuth is a number ranging from 0 to 360 which describes a horizontal direction in 3 dimensional space relative to the used coordinate system.

- Course is a number ranging from 0 to 360 which describes your true direction of movement relative to a point of reference which is usually either true North or magnetic North. 
When we say the direction is 'true' it needs to be true in relation to something, which is usually ground but sometimes water or air. Note that "true direction" has nothing to do with "true North" but can be expressed in relation to true North.

- Heading or orientation (I think these are the same thing) describe which direction you are facing, but not which way you are actually moving.

- Track describes the actual path you took (where you came from). A track describes the previous courses (this can be multiple courses).

- Route describes the intended path of travel. A path can consist of multiple bearings.

From what I gather Bearing might not be the right word, it Course might be more accurate.

barbeau, 2019-08-18:
Thanks for sharing! The current labels in GPSTest are a 1:1 match to the names used in the Android Location API where the data comes from:
https://developer.android.com/reference/android/location/Location.html

I'd like to keep it this way to make it clear where the data is coming from.

closed 2019-08-18 by barbeau

Meteor0id, 2019-08-18:
Right, why would we critically think about the information we're spreading if uncle Google has already voiced their opinion?

I disagree with that reasoning. This app specializes in GNSS information, and can set an example and can serve an educational purpose. If we just follow oversimplified terminology made up by Google the app is actually misinforming users in some cases.

And hadn't you deliberately added extra flags and extra information to warn users that the API isn't clear on some points (such as accuracy and SBAS usage)? Doesn't that also show that the API simply isn't at the level we want this app to be, and that providing extra information is worth doing?

barbeau, 2019-08-18:
You're welcome to disagree :). 

GPSTest provides a very unfiltered view into what the OEM is implementing on the device for GNSS capabilities. I want to highlight when things aren't clear, but don't want to alter the information coming out of the device. Otherwise when a developer from the OEM looks at the app it's not clear to them where the data is coming from, and makes it harder for them to fix problems.

Meteor0id, 2019-08-19:
What you write makes sense, but on the other side if anyone looks at the app and the terminology isn't correct, that is also likely to lead to situations where it also isn't clear why the information in the app doesn't correspond to the expected value. Any  developer can look at the code, but anyone who doesn't notice the indiscrepancy in the terminology has no chance within the app itself to notice that.