barbeau/gpstest #336: GNSS frequencies don't have a nice name

TaaviE, 2019-08-20:
**Summary:** 

A few GNSS frequencies don't have a nice name.

**Steps to reproduce:** 

Open GPSTest on a Mi 9

**Expected behavior:** 

What did you expect the app to do?

**Observed behavior:** 

The nice names aren't displayed for these frequencies:
* GALILEO 1602.000 MHz , ID 7, 19
* GALILEO 1561.098 MHz, ID 7, 19, 26, 31, 31
* BEIDOU 1602.000 MHz, ID 2, 5, 9, 13, 16
* Galileo SBAS 1602.000 MHz, ID 126, 136

**Device and Android version:** 

Mi 9, 9

barbeau, 2019-08-20:
@TaaviE Thanks for reporting this!  Could you include a screenshot too?

Unfortunately it looks like these are bugs - as far as I can tell there aren't any real signals being broadcast at these frequencies. Here's an example GNSS frequency chart:

[image]

Could you please report this bug to Xiaomi and see what they say?

TaaviE, 2019-08-20:
I can try asking Xiaomi to fix these.

[image]

PedroVenancio, 2019-08-21:
Hi @barbeau and @TaaviE 

I also see several strange things in my Xiaomi Mi9, with Android 9 PKQ1.181121.001 and MIUI Global 10.2.13 (GPSTest v3.4.0 and v3.4.2).

[image]

[image]

1) Several GPS satellites repeated: ID 24 and ID 30 on L1;

2) GLONASS L5 label (is this supported?);

3) Galileu frequencies of 1561.098 MHz; 

4) BeiDou frequencies of 1602.000 MHz;

5) Galileu SBAS ID 136, with frequencies of 1561.098 MHz.

By the way, is SBAS supported by Mi 9? 

I must say that Mi 9 is a very disappointing device to me in terms of GNSS. It is a huge drawback when compared to Mi 8... The unsupported accumulated delta range data is the worst thing...

@TaaviE  What channels do you use to communicate with Xiaomi?

Thanks!

TaaviE, 2019-08-21:
> Several GPS satellites repeated: ID 24 and ID 30 on L1;

I think some satellites do send two signals on two bands?

> Galileu frequencies of 1561.098 MHz;
> BeiDou frequencies of 1602.000 MHz;

I think these are in wrong order because the code that deals with the second GNSS bands has basically a small typo.

> By the way, is SBAS supported by Mi 9?

The support should exist on all Snapdragon 855 devices.

> I must say that Mi 9 is a very disappointing device to me in terms of GNSS.

The small typo that is in a library somewhere isn't that big of a problem honestly. I can get a GPS fix inside quite a few buildings and the accuracy outside is 2.5m easily 99.99% of the time when I've recorded long trips.

> The unsupported accumulated delta range data is the worst thing...

I'm curious, what can you do with that data?

> What channels do you use to communicate with Xiaomi?

The built-in "Feedback" application in settings. If enough people report a problem then they fix it.

PedroVenancio, 2019-08-21:
Thanks for your feedback @TaaviE !

> 
> > The unsupported accumulated delta range data is the worst thing...
> 
> I'm curious, what can you do with that data? Also have you tried the MIUI beta maybe it exists there?
> 

Accumulated Delta Range is necessary for post-processing. Please, see this issue https://github.com/barbeau/gpstest/issues/268

TaaviE, 2019-08-21:
> Accumulated Delta Range is necessary for post-processing. 

I found that information but I haven't found what (error) it corrects?

PedroVenancio, 2019-08-21:
> 
> > Accumulated Delta Range is necessary for post-processing.
> 
> I found that information but I haven't found what (error) it corrects?

https://developer.android.com/reference/android/location/GnssMeasurement.html#getAccumulatedDeltaRangeMeters() 

https://insidegnss.com/generating-carrier-phase-measurements/

benzfish, 2019-09-02:
Mi9 shows both L1 and L5 at the column L1, its a bug. 
Xiaomi will fix this bug in the next release.

mimichris, 2019-10-13:
Hello,
On the latest version MI9, MIUI 10.2.14 of this month, no change, I wonder if they will really change something ....

TaaviE, 2019-10-13:
They fixed some but not GALILEO, I submitted an error report, please do the
same and ask others as well.

Bolognesus, 2019-10-26:
[image]
Mi 9T pro, same issue.

DominicHayes, 2019-10-29:
[image]
Impressed by the performance even at 10000m and 800kph, but the labels still need attention. 9T Pro.

barbeau, 2019-10-29:
@DominicHayes Thanks for the screenshot! The carrier frequency of 1602 MHz for Galileo and Beidou is an error by Xiaomi - there are no real GNSS signals broadcast on these frequencies from these systems. Could you please report this bug to Xiaomi and see what they say? 

Has anyone else gotten a response from Xiaomi on this recently?

TaaviE, 2019-10-29:
@DominicHayes 
[image]

I got 52 sattelites on almost the same altitude with a Mi 9, also has a few weird Galileo frequencies.

I'm waiting for MIUI11 to see if Xiaomi has fixed it there.

PedroVenancio, 2019-11-04:
I've just updated to MIUI 11 and it seems that this issue was fixes:

[image]

TaaviE, 2019-11-05:
Did the same, seems fixed indeed.

closed 2019-11-05 by TaaviE

mimichris, 2019-11-05:
Hello,
I do not have the possibility of this update yet, when did you have it?
Thank you.

PedroVenancio, 2019-11-05:
> 
> 
> Hello,
> I do not have the possibility of this update yet, when did you have it?
> Thank you.

@mimichris  It landed in my Mi9 last night.

mimichris, 2019-11-05:
Hello,Maybe I had not looked at the possibility of updating, I just made this update and I no longer have the satellite data error.
mimichris

alpter, 2019-12-10:
Did you see this ? 
[https://insidegnss.com/galileo-hits-the-spot]()