barbeau/gpstest #352: String to be translated

Giolly90, 2019-10-16:
**Summary:** 
When entering transifex I don't see the string showed in the screenshot

**Steps to reproduce:** 
Enter Accuracy view, tap on QR code icon, tap OK

**Expected behavior:** 
I see a message in my language

**Observed behavior:** 
I see a message in english

**Device and Android version:** 
OnePlus 5, Android 9

**Screenshots:** 
[image]

Giolly90, 2019-10-16:
Moreover, choice buttons YES/NO are not well defined

barbeau, 2019-10-16:
@Giolly90 thanks for flagging this! Right now this screen is controlled by the Bar Code Scanner app library, so I don't have direct control over localizing it or fixing the buttons. I'll leave this issue open for improving this in the future and possibly replacing it with code in GPSTest.

goldfndr, 2019-10-19:
[Relevant code](https://github.com/zxing/zxing/blob/master/android-integration/src/main/java/com/google/zxing/integration/android/IntentIntegrator.java#L115-L117)

paulle, 2019-10-20:
Why not making the yes/no-buttons round, so they will be better separated.

barbeau, 2019-10-20:
The yes and no buttons should follow the same normal material design dialog layout as the other dialogs in the app.

paulle, 2019-10-20:
I understand, so the buttons should only be a little bit more separated.

closed 2022-04-11 by stale[bot]