barbeau/gpstest #356: GLONASS L1 and L2 frequency ranges for labels too wide

barbeau, 2019-11-13:
**Summary:** 

First raised in https://github.com/barbeau/gpstest/issues/103#issuecomment-552140034.

Our frequency ranges for GLONASS L1 and L2 labels are slightly too large. AFAIK this doesn't have any affect on end users as no other satellites occupy the overlapped frequencies, but we should fix this anyway.

ESA Navipedia states the correct frequency ranges for L1 and L2:
https://gssc.esa.int/navipedia/index.php/GLONASS_Signal_Plan

[image]

[image]

New frequencies should be:

~~~
                if (carrierFrequencyMhz >= 1598.0625f && carrierFrequencyMhz <= 1605.375f) {
                    return "L1";
                } else if (carrierFrequencyMhz >= 1242.9375f && carrierFrequencyMhz <= 1248.625f) {
                    return "L2";
                }
~~~

**Steps to reproduce:** 

Review GLONASS L1 and L2 frequencies used in code to label these frequencies

**Expected behavior:** 

Use exact frequency range

**Observed behavior:** 

A slightly larger frequency band than the actual used frequencies is used

goldfndr, 2019-11-13:
A question: would it be useful to display an offset? (e.g. for a frequency of 1605.375 display something like "L1+6" or "L1⁺⁶" or "L1₊₆"? It could even be a superscript for + and subscript for -, but that might not be very intuitive.)

barbeau, 2019-11-14:
@goldfndr I'd like to display this somewhere eventually, although I agree that trying to fit this into the main CF label probably isn't going to be intuitive. This could be shown when tapping on a satellite for more detail - I'll add this to https://github.com/barbeau/gpstest/issues/113.

commit 4e9e61ce (2019-11-14): Fix #356 - Narrow GLONASS frequency bands for L1 and L2
closed 2019-11-14 by barbeau