barbeau/gpstest #357: Blinding white at startup with dark theme used

goldfndr, 2019-12-08:
**Summary:** 

Launching GPSTest always results in a white screen even with dark theme enabled.

**Steps to reproduce:** 

Launch the app. (Version 3.5.2)

**Expected behavior:** 

Either launch with themed background or delay painting of background until theme loads.

**Observed behavior:** 

Launched with white background. While it might be brief if it is the only app launched, launching successive apps via Tasker (OsmAnd, WiGLE WiFi Wardriving, GPSTest) prolongs the blinding white while the apps pursue their background activity.

**Device and Android version:** 

Samsung Galaxy Note 4, Android 6.0 Marshmallow stock, Google Play Services 19.6.29

barbeau, 2019-12-10:
Thanks @goldfndr. I think this should be resolved when switching to official dark mode support as discussed in https://github.com/barbeau/gpstest/issues/277.

barbeau, 2021-09-23:
I'm closing this as it should be covered by https://github.com/barbeau/gpstest/issues/277.

closed 2021-09-23 by barbeau