barbeau/gpstest #358: Add 24-hour format

mirabilos, 2019-12-13:
**Summary:** 

I can either use gpstest in English with the very foreign AM/PM format, or in German (which is only half-translated and I don’t like localisation in general as it irritates me, **and** it breaks coordinate format) with 24-hour clock format, but not English/International with 24-hour clock format.

**Steps to reproduce:** 

Select English language in the menu, see AM/PM format clock ☹

**Expected behavior:** 

Have a toggle for clock format, like most äpps have.

**Observed behavior:** 

AM/PM format

**Device and Android version:** 

HTC Desire with LiquidSmooth 3.2 (AOSP KitKat / Android 4.4), µG (microG) installed.

**Screenshots:** 

Do you really need one?

barbeau, 2019-12-17:
Thanks @mirabilos, I agree this would be a nice option to have. BTW, if you have any free time and would like to help update the German translation, please see https://github.com/barbeau/gpstest/blob/master/TRANSLATIONS.md!

goldfndr, 2020-01-18:
See also #117.

mirabilos, 2021-06-03:
?

mirabilos, 2024-01-17:
stale[bot] dixit:

>This issue has been automatically marked as stale because it has not

stalebot considered harmful

barbeau, 2025-02-22:
Not stale