barbeau/gpstest #359: Localisation breaks coordinate format

mirabilos, 2019-12-13:
**Summary:** 

Localised decimal separators are mistakenly used in coordinate format.

**Steps to reproduce:** 

Select decimal minutes (geocaching) coordinate format, then switch to German language.

**Expected behavior:** 

Coordinates are still in `N 50° 42.000  E 007° 07.000` style.

**Observed behavior:** 

Coordinates have a comma in them.

**Device and Android version:** 

HTC Desire, LiquidSmooth 3.2 (AOSP KitKat / Android 4.4), µG (microG)

**Screenshots:** 

Do you really need one?

barbeau, 2019-12-18:
@mirabilos So this was actually a deliberate choice to fully support localization in other languages. I'm open to hearing about this from more users, but my current understanding is that latitude and longitude are typically localized just like any other decimal number.

Is there a particular use case that this affects? For example, copying/pasting into other software?

mirabilos, 2019-12-18:
Sean Barbeau dixit:

>@mirabilos So this was actually a deliberate choice to fully support
>@localization in other languages. I'm open to hearing about this from
>@more users, but my current understanding is that latitude and
>@longitude are typically localized just like any other decimal number.

Hm, interesting, because this goes completely contrary to my
understanding and experience from GPS Stash Hunting (geocaching)
and other GPS-based activities.

It is even considered wrong to localise the N/S and E/W parts
(for instance, “E” would be “O” in German, but nobody does this
any more in the GPS era — this is only found on ancient pre-Blue-
Switch maps).

>Is there a particular use case that this affects? For example,
>copying/pasting into other software?

I’d assume some software might be affected; I’ve not tested this.
My initial report was based on æsthetics for myself.

bye,
//mirabilos
-- 
I want one of these. They cost 720 € though… good they don’t have the HD hole,
which indicates 3½″ floppies with double capacity… still. A tad too much, atm.
‣ http://www.floppytable.com/floppytable-images-1.html

barbeau, 2020-01-13:
I'm going to close this out for now to clean up some issues.

closed 2020-01-13 by barbeau