barbeau/gpstest #361: # Sats  technically not correct anymore with dual GPS

schwabe, 2020-01-10:
So this might be as well a wontfix. Before dual frequency number of satellites with lock was the same as number of frequencies with lock. With dual frequency that is no longer true.

I am not sure how to really improve on that. Maybe add a whole new section of stats? Like # sats with dual frequency, rename # Sats just to # Locks?

barbeau, 2020-01-10:
@schwabe Thanks for pointing this out! Yes, I'm tracking this issue in https://github.com/barbeau/gpstest/issues/343, so I'm going to close this one as a duplicate.

To fix this field I need to implement grouping of satellites according to constellation, ID, and carrier frequencies, and this is almost done in https://github.com/barbeau/gpstest/pull/355.

barbeau, 2020-01-13:
I'm actually going to re-open this issue and use it to specifically track fixing the "# Sats" label on dual-frequency devices. Showing both # Sats and #Signals requires a little more UI work so I'll continue tracking that in https://github.com/barbeau/gpstest/issues/343.

closed 2020-01-23 by barbeau

barbeau, 2020-01-24:
Number of satellites should now be correct for dual-frequency devices in the new beta release v3.5.3 - https://play.google.com/apps/testing/com.android.gpstest.

If you have a dual-frequency device please test and let me know if it looks right!

commit 174d25ee (2020-01-23): Fix #361 - Group signals into satellites for correct dual-frequency used/in view counts (#355)

* Includes flagging duplicate constellation/ID/carrier signals, as well as unknown carrier frequencies.
* Includes detecting if dual-frequency signals are in view and in use
* Move most of the non-satellite/signal utilities out of GPSTestUtil and rename it to SatelliteUtils
* Convert carrier freq utils from Java to Kotlin
* Bump test runner dependency
* Includes unit tests