barbeau/gpstest #363: Available accuracy/error level in Android?

Jon8RFC, 2020-01-13:
Sean, I tried to find you on XDA to ask this question there, but couldn't find a profile.

What are your observations and knowledge on Android's inherent accuracy or error level of positioning?  I ask because of this bug report I've posted:
https://issuetracker.google.com/issues/146010208

Unfortunately I've only used your application in the past 2-3 years.  I've used GPS Status & Toolbox since 2009, and had 6ft accuracy, followed by 9.9ft and now 12.4ft.  How likely is it that Android has created this restriction, versus the hardware itself actually getting worse over the years, such as a smaller antenna?  I'm second-guessing myself, but with how quickly cold starts acquire signals these days, I can't imagine the antenna itself getting smaller/worse over the years.  After seeing that external receivers get sub-3meter accuracy, I can't blame it on newer satellites using "cheap" hardware compared to the decommissioned ones.

Also, as a bit of other information, I've read that GLONASS "[wasn't allowed/supported](https://www.gpsworld.com/us-still-not-allowing-glonass-stations/)" by consumer devices in the USA until recent years, just how GALILEO was recently permitted in many with a software update.  I don't recall if my 2009 Motorola Droid used GLONASS, but by Motorola Droid 2 Global (Verizon Wireless-branded and bootloader locked, until exploited) absolutely did use GLONASS back in February of 2011 with Cyanogenmod, as did every phone I used cyanogenmod on after that, even if phone specifications explicitly excluded GLONASS support.  It's apparent that there are software restrictions for GPS usage.

What's your take on all of this?  Am I crazy, or has Android truly been nerfing accuracy, a la Selective Availability, over time?

barbeau, 2020-01-16:
@Jon8RFC Good question! So it's important to remember that the "accuracy" value you see in the app is just an estimate by the hardware OEM. Here's an article with more details on what that means - https://medium.com/@sjbarbeau/measuring-gnss-accuracy-on-android-devices-6824492a1389.

The true error/accuracy is unknown to the hardware and needs to be measured, which you can do with the GPSTest "Accuracy" feature.

Your Android bug report is interesting - it means that the Pixel 4 hardware is effectively capping the estimated horizontal accuracy at 12.4ft/3.8m. I'm guessing that it's a function of the algorithm being used to estimate accuracy, which makes sense, as this implementation probably changed in the switch to dual-frequency. So the algorithm is effectively refusing to say that it can be more than 68% confident that the position it has calculated lies within the 3.8 meter radius - but that doesn't mean that the actual accuracy isn't better than 3.8 meters.

Because these estimated accuracy values come from the hardware OEM, they can differ widely across devices, and really depend on the algorithm the OEM is using to estimate accuracy. So if this number is trending up over time, it's possible that OEMs are just becoming more conservative in their estimates (not that actual accuracy is getting worse). But, the only way we will know this is if people start measuring/reporting GNSS accuracy.

(FYI, for future reference, the Google Group https://groups.google.com/forum/#!forum/gpstest_android is probably a better place for this type of question)

closed 2020-01-16 by barbeau

Jon8RFC, 2020-01-17:
Thanks for the great information and the links!