barbeau/gpstest #364: "E Acc" should be "E H Acc"

goldfndr, 2020-01-18:
**Summary:** 

On older phones, the first Accuracy label should mention that it's for horizontal accuracy.

**Steps to reproduce:** 

Launch app on a pre-7.0 (API 23 or earlier) device, switch to Status view, see label.

**Expected behavior:** 

Display a label indicating horizontal accuracy, "E H Acc".

**Observed behavior:** 

[As mentioned in the help](https://github.com/barbeau/gpstest/blob/master/GPSTest/src/main/res/values/strings.xml#L336), the "E Acc" value is for estimated horizontal accuracy. On newer phones, this instead has a "E H/V Acc" label+value (according to the screenshots I've seen), and I think older phones could benefit from a more meaningful label.

(Also, [the accuracy labels](https://github.com/barbeau/gpstest/blob/master/GPSTest/src/main/res/values/strings.xml#L74-L79) should presumably be harmonized on whether [they have an "E"](https://github.com/barbeau/gpstest/issues/320#issuecomment-521322276) or period, but I don't see those on my phone.)

**Device and Android version:** 

Samsung Galaxy Note 4, Android 6 stock

barbeau, 2020-01-21:
Thanks @goldfndr, that makes sense. IIRC I dropped the period on "E H/V Acc" due to length restrictions, but I'll keep it on the others.

closed 2020-01-21 by barbeau

goldfndr, 2020-01-23:
I'm confused: why aren't you adding an "E " to the others or a period to this one?

barbeau, 2020-01-23:
I'm not adding the periods because on devices with smaller screen widths it will push the value off screen. The main reason I've added "E" to accuracy is to differentiate between estimated and actual (measured) accuracy - many users confuse these. We don't have a way to easily measure true speed or bearing accuracy within the app, so I don't think we need to stress the difference there.

commit cadd5ca9 (2020-01-21): Fix #364 - Indicate that estimated accuracy is horizontal (legacy devices)