barbeau/gpstest #377: map view improvments

mash-graz, 2020-02-21:
**Summary:**
 
* map view does not show any actual location marker
 
and 

* orthofoto/satellite and hybrid view does not work in the map view and accuracy section.
the accuracy tab is therefore nearly unusable.

**Steps to reproduce:** 

the missing location marker in map view mode should be reproducable on any device, but the second issue could be related to my mobile operating system ([/e/](https://e.foundation/)), which tries to block and replace all google services by free alternatives.

**Expected behavior:** 

it would be nice, if the application would support a few other map providers.

**Device and Android version:** 

Xiaomi Mi 8

/e/ 0.7  -- based on Android Pie -- microG 0.2.10

mash-graz, 2020-02-25:
the reported issue could be very likely caused by flaws in the microG map viewer, because there are similar effects also noticeable in other apps on the same device.

barbeau, 2020-02-25:
@mash-graz Have you tried the OSM Droid build of GPSTest? https://f-droid.org/packages/com.android.gpstest.osmdroid/

It's probably more reliable than the Google build using microG, and unfortunately I can't really fix microG-specific issues in this build.

Note that on the OSMDroid there is a known issues on the Accuracy map view for tiles not appearing - see https://github.com/barbeau/gpstest/issues/316.

closed 2020-02-25 by barbeau