barbeau/gpstest #378: Wrong time shown

mirabilos, 2020-02-21:
**Summary:** 

The “Time:” field contains data off by an hour.

**Steps to reproduce:** 

Let it get a fix; compare “Time:” shown with localtime (in Android system bar) and UTC.

**Expected behavior:** 

“Time:” shows localtime as configured in Android (identical with system bar), _or_ UTC.

**Observed behavior:** 

“Time:” shows time off by an hour, e.g. it is now 15:05 CET (UTC+01:00), that is 14:05 UTC, but “Time:” shows 04:05:14 PM which, if I parse this American format right, is 16:05, i.e. Eastern European Time.

**Device and Android version:** 

HTC Desire “bravo”, LiquidSmooth 3.2 (AOSP KitKat 4.4), µG

**Screenshots:** 

Can be created by pressing the Volume Down and Power Button at the same time on Android 4.0 and higher. If needed, I’ll create and attach.

barbeau, 2020-02-21:
@mirabilos Can you please enroll in the beta testing group (https://play.google.com/apps/testing/com.android.gpstest) and try v3.6.0, which should be rolling out shortly?

I'm guessing this device might have the GPS week rollover issue. v3.6.0 highlights times that are wrong due to GPS week rollover in red, and you can tap on them for more details, including a readout off the full date and time.

Please give this a try and let me know what you see - screenshots would be great.  Thanks!

mirabilos, 2020-02-22:
Sean Barbeau dixit:

>@mirabilos Can you please enroll in the beta testing group
>@(https://play.google.com/apps/testing/com.android.gpstest) and try

No.

JonnyTech, 2020-02-22:
@barbeau for those of us non-google users how can we get the beta?

barbeau, 2020-02-22:
@JonnyTech OSMdroid should pick it up and publish it, if you want to download the APK directly see:
https://github.com/barbeau/gpstest/releases/tag/v3.6.0

barbeau, 2020-02-25:
@mirabilos Can you please try the v3.6.0 build of the OSMDroid release and let me know if you see the red highlighting around your time field?

https://github.com/barbeau/gpstest/releases/tag/v3.6.0

barbeau, 2020-03-13:
I'm closing this out. If I can get more info about this issue and it doesn't seem to be an instance of GPS week rollover, then we can revisit.

closed 2020-03-13 by barbeau