barbeau/gpstest #383: No BeiDou and Galileo on Mi8

szangvil, 2020-02-26:
**Summary:** 

Xiaomi Mi8 is not using BeiDou and Galileo

**Steps to reproduce:** 

Just open the app

**Expected behavior:** 

Track all 4 constalltions

**Observed behavior:** 

A few days ago when I was Android 9, I would see all 4 constellations on the Mi8. Out of the blue and for no apparent reason, the device started to use GPS and Glonass only. Several reboots did not help.
I upgraded to Android 10 only to get the same results.

In the screenshot, you can see OnePlus 7 and RealMe X2 tracking all four constellations while the Mi8 only tracks GPS L1, L5, and Glonass L1.

A second Mi8 behaves the same way.

**Device and Android version:** 

Mi8, Android 10

**Screenshots:** 

[image]

szangvil, 2020-02-26:
The picture below shows the versions of each phone:
[image]

barbeau, 2020-02-26:
@szangvil Interesting - have you tried contacting Xiaomi and see what they say? Were you in roughly the same physical location when you previously got BeiDou and Galileo, compared to now when you don't?

There isn't anything GPSTest can do to change this, but I'm interested to know if others have seen something similar.

szangvil, 2020-02-26:
@barbeau, as you can see in the attached photo, all three phones (OnePlus 7, Realme X2 and the Mi8) are on my desk getting very good signals from a GNSS repeater we have in our office.

It might be an issue with Android 10 on the Mi8... 

I did not contact Xiaomi yet...

szangvil, 2020-02-27:
Update: Seems like I am not the only one...

https://c.mi.com/thread-2880216-1-1.html

barbeau, 2020-03-13:
@szangvil Any fix for this from Xiaomi, or is it still ongoing?

szangvil, 2020-03-15:
No fix at all.

After a phone data wipe I started getting all 4 constellations but after a few days, it went back to the same state (only GPS and GAL).

barbeau, 2020-03-15:
@szangvil interesting. Did the device install an update prior to Beidou and Galileo disappearing again?

Can you try running this via ADB and seeing if it returns anything?

`adb shell settings get global gnss_satellite_blacklist`

See https://groups.google.com/d/msg/gpstest_android/KrBe_csYFfg/OJ5fmNcWAQAJ for details - Xiaomi may be blacklisting satellites via this setting.

barbeau, 2020-04-29:
Closing this for issue clean-up, as it's a device issue and not the app. But, please continue to post updates if anyone has more info.

closed 2020-04-29 by barbeau