barbeau/gpstest #384: Show GNSS time in default device locale

barbeau, 2020-02-27:
**Summary:** 

Currently GNSS time is formatted as follows:

~~~
    SimpleDateFormat mDateFormat = new SimpleDateFormat(
            DateFormat.is24HourFormat(Application.get().getApplicationContext())
                    ? "HH:mm:ss" : "hh:mm:ss a");
~~~

It should be formatted in the default locale.

Note to myself - I need to look at #117 for the history of this.

Related to request for Settings to toggle 24 hours format - https://github.com/barbeau/gpstest/issues/358.

**Steps to reproduce:** 

Look at time in Status

**Expected behavior:** 

Show time in my default locale

**Observed behavior:** 

Time is formatted as above

**Device and Android version:** 

N/A

ValZapod, 2020-06-28:
Why the time has GMT+03:00 or whatever? It better be UTC+03:00. GMT zone now is defined as UTC not as GMT time standard.

barbeau, 2020-06-30:
@ValZapod Could you share a screenshot of what you're seeing?

ValZapod, 2020-06-30:
[image]

barbeau, 2020-07-01:
@ValZapod What time zone are you located in?

ValZapod, 2020-07-01:
MSK UTC+03:00, Moscow.

barbeau, 2020-07-07:
@ValZapod Thanks! In short, I'm not exactly sure why Android prefers the text `GMT` over `UTC` - it's the output of the `SimpleDateFormat` class in Android:
https://developer.android.com/reference/java/text/SimpleDateFormat#timezone

For me in Tampa, FL, USA, it shows `Jul 7, 2020 EDT`.

barbeau, 2020-07-07:
Looking at this issue again, I believe it can be closed - we already have the time and date based on device locale. See https://github.com/barbeau/gpstest/issues/117#issuecomment-368720692.

closed 2020-07-07 by barbeau

ValZapod, 2020-07-07:
> For me in Tampa, FL, USA, it shows `Jul 7, 2020 EDT`.

Because from that doc 
>SimpleDateFormat also supports localized date and time pattern strings.