barbeau/gpstest #387: Forcing xtra data injection

aleasto, 2020-03-06:
`force_xtra_injection` got renamed to `force_psds_data` in Android 10

barbeau, 2020-03-11:
Thanks for the heads up @aleasto! I'll work on adapting the app for this.

barbeau, 2020-03-11:
Confirmed that this is new in Android 10 - the last tag of Android 9 still uses `force_xtra_injection`:
https://android.googlesource.com/platform/frameworks/base.git/+/refs/tags/android-9.0.0_r54/services/core/java/com/android/server/location/GnssLocationProvider.java#1451

barbeau, 2020-03-13:
And here's the commit that changed it:
https://android.googlesource.com/platform/frameworks/base.git/+/8c6ac67a9657777b7f1d14f13cd2747f247185da

PSDS = Predicted Satellite Data Service

closed 2020-03-13 by barbeau

barbeau, 2020-03-14:
Released in v3.6.3, currently out in beta on Google Play, if anyone with an Android 10 device that supports PSDS wants to test:
https://groups.google.com/forum/#!topic/gpstest_android/LulZpdhpHpM

commit efb3b520 (2020-03-13): Fix #387 - Use force_psds_injection instead of force_xtra_injection on Android 10 and higher

* Rename XTRA to PSDS on all Android versions for consistency and simplicity in translations