barbeau/gpstest #388: How to check if there is dophler effect support by the device?

avkonst, 2020-03-27:
**Summary:** 

Just a question. Want to review how accurate speed measuments by my phone.

**Expected behavior:** 

Would like to see if the devices measures the speed using dophler effect or not.

**Device and Android version:** 

Armor X5, android 9

barbeau, 2020-03-29:
Unfortunately Android doesn't provide any indication for how speed is calculated. You can look at the S. Acc field to get some idea of speed accuracy (https://developer.android.com/reference/android/location/Location#getSpeedAccuracyMetersPerSecond()):

>We define speed accuracy at 68% confidence. Specifically, as 1-side of the 2-sided range above and below the estimated speed reported by getSpeed(), within which there is a 68% probability of finding the true speed.

closed 2020-03-29 by barbeau

avkonst, 2020-03-29:
Thanks for the answer. Very nice. I can benchmark 2 phones now. One is clearly better. I am going to compare my best mobile phone with the [gps speed watches](https://www.locosystech.com/en/product/gps-watch-gw-60.html) which are used during [Luderitz windsurfing speed challenge](https://www.youtube.com/watch?v=O4G2EKwaiVE). It will take some time with this virus lockdown. I can post here the update if you are interested in.

barbeau, 2020-03-29:
Yes, please share results!