barbeau/gpstest #391: Galileo and BeiDou satelites disappear in 25 second after start

Losinsky, 2020-04-23:
Galileo and BeiDou satelites disappear in 25 second after start everytime I start application. GPS and GLONASS satelites remain and their data update continuously.
You can see at attached screencast Galileo and BeiDou disappeared in 27 second from start.

**Observed behavior:** 
[Screencast_2020-04-16-17-28-52.zip](https://github.com/barbeau/gpstest/files/4525632/Screencast_2020-04-16-17-28-52.zip)

**Device and Android version:** 
BQ Aquaris X, Android 8.1.0 stock
Google Play Services 20.12.16

barbeau, 2020-04-29:
@Losinsky Thanks for reporting this! That's really strange. Another user reported a similar issue with a BQ Aquaris device. 

It's definitely an issue with the device and not the app, so I'd suggest you contact BQ and ask why this is happening. Feel free to post on the Google Group about this too so other users can chime in with what they are seeing too:
https://groups.google.com/forum/#!forum/gpstest_android

closed 2020-04-29 by barbeau

barbeau, 2020-04-29:
See also related issue https://github.com/barbeau/gpstest/issues/383 for the Xiaomi Mi 8.

@Losinsky Can you try running the below command via ADB and seeing if it returns anything, both before and after the satellites disappear?

`adb shell settings get global gnss_satellite_blacklist`

See https://groups.google.com/d/msg/gpstest_android/KrBe_csYFfg/OJ5fmNcWAQAJ for details - Xiaomi may be blacklisting satellites via this setting.

Giolly90, 2020-04-29:
Hi, I do have similar instability with BeiDou and Galileo satellites. In my case, when I start the app I see immediately a certain number of GPS, Glonass and BeiDou satellites even if they have no C/N0 values, it is like they are memorized by the app or device.
For what concerns Galileo satellites, they only appear after let's say 2 minutes after app opening, and I see only those satellites the device can actually see.
Often BeiDou and Galileo satellites disappear all suddenly and I have to kill the app and start it again.

Device: OnePlus 5 so Android 9 stock