barbeau/gpstest #393: Numbers do not fit fully on the screen

jidanni, 2020-04-28:
The only thing I might be guilty of here on my Zenfone 3, with Android 8.
8 years ago having selected song larger font in Android settings do the being 60 years old. Well, as you can see the numbers are cut off at the top oh, and as azimuth is all stretched out at the bottom. When, all along, there's still enough room to move things around and squeeze things in so they don't get cut off or made  vertical etc etc etc.
[image]

jidanni, 2020-04-28:
First, on the bottom, why don't you let the title line render by itself, and then the rest of the table render by itself also.With independent rendering the wider title line will not ruin the following flag lines.

jidanni, 2020-04-28:
Now, for the top part of the image, being that the right rows are not in a perfect column already, there is not any reason not to let them slide over even more enabling everybody to see the remaining words all the way up to the end of each line.

jidanni, 2020-04-28:
Sure, I can just rotate the screen to see everything in  landscape mode, but that really shouldn't be necessary. You've got it 99% correct already.

jidanni, 2020-04-28:
And, for the bottom part of the screen, you might as well put the title bar into the blue Status area (very top of window), because once the user scrolls down he won't see it anyway.
Hope but if he has a school down yet then it would be out of place. Nevermind.

barbeau, 2020-04-28:
@jidanni Thanks for the feedback! I'm closing this as a duplicate of https://github.com/barbeau/gpstest/issues/176. Please comment there so we can keep all related comments in the same place.

closed 2020-04-28 by barbeau