barbeau/gpstest #394: openstreetmap

jidanni, 2020-04-28:
Allow the user to at least choose one type of openstreetmap, instead of the several types of proprietary maps.

barbeau, 2020-04-28:
@jidanni Thanks for the feedback! There is a pure OSS version on F-Droid with OSM maps you can check out:
https://f-droid.org/packages/com.android.gpstest.osmdroid/

closed 2020-04-28 by barbeau

jidanni, 2020-04-29:
Waaa... One has to install/use F-droid (uggg!) to get a version that uses openstreetmap (but then probably not googlemaps,)
I have an idea. Do like https://play.google.com/store/apps/details?id=net.psyberia.offlinemaps , where one can not only choose google maps, but also openstreetmaps!

barbeau, 2020-04-29:
No need to install FDroid - you can download the APK from the releases page:
https://github.com/barbeau/gpstest/releases

jidanni, 2020-04-29:
I'm telling you, it is entirely legal to offer openstreetmap and google
maps on the same menu. You needn't maintain two entirely separate
versions of the app to do this.

Google will not kick you off Google Play for combining your two apps
into a single version.

Yes, maybe in the past they would, but not anymore.

jidanni, 2020-04-30:
As an example, one of my houses is deep in the woods. All you can see on Google is trees therefore I can't use Google. I need openstreetmap because it's on openstreetmap.

Another one of my houses is clearly visible on Google because there's no trees on top of it. As you can see it would be great to have both Google and openstreetmap in the same app.