barbeau/gpstest #395: Map needs scale bar

jidanni, 2020-04-28:
The map already has a North Arrow. Now all that it needs to be a standard map is a scale bar.

barbeau, 2020-04-29:
@jidanni Thanks for the feedback!

Looks like OSMDroid supports this natively:
https://osmdroid.github.io/osmdroid/javadocAll/org/osmdroid/views/overlay/ScaleBarOverlay.html

...but to my knowledge Google Maps does not.

There are also third-party libraries to achieve this:
https://github.com/pengrad/MapScaleView

It may be easiest to just use the 3rd party library on both for consistency.

jidanni, 2020-04-29:
I also found
https://stackoverflow.com/questions/2094648/google-maps-api-how-to-add-a-scale-measure-on-a-map

barbeau, 2020-05-27:
I've opened an issue on the Android Maps Utils library to support this feature too - https://github.com/googlemaps/android-maps-utils/issues/701.

barbeau, 2023-12-13:
Not stale

barbeau, 2025-01-18:
Not stale