barbeau/gpstest #396: On-off button  could be clearer

jidanni, 2020-04-28:
Okay, I finally figured out what that switch at the top of the screen does.
But consider using a more familiar on-off button. One that becomes a square or a circle.
Currently just looking at it it's hard to tell which position is on and which position is off.

jidanni, 2020-04-28:
An even bigger problem is it moves around depending on which page you're looking at.
Sometimes it's at the most right part of the screen, but sometimes it gets shoved around by a share button or a menu button. Please be consistent in where you put it.

barbeau, 2020-04-28:
@jidanni Thanks for the feedback! Closing as duplicate of https://github.com/barbeau/gpstest/issues/325.

closed 2020-04-28 by barbeau