barbeau/gpstest #400: Show GPS uptime on status page

jidanni, 2020-04-28:
On the Status page, it seems that there's just one item missing: how much time has the GPS been turned on for.

barbeau, 2020-04-29:
@jidanni Thanks for the feedback! Android doesn't provide this value directly, so the app would need to calculate it, which poses some challenges depending on what you're looking for.

Some questions:
1. What would you use the uptime value for?
1. Do you expect the app to show (a) how long it has been running for, or (b) how long the underlying GNSS hardware has been active?

(a) is pretty straight forward, but (b) is harder as the GNSS system typically duty-cycles very frequently to conserve energy. Without looking closer at actual device behavior with the callbacks, it's not clear if the app would receive all the information needed to determine actual hardware uptime (i.e., how long the GNSS hardware has been powered on). The app also doesn't know if the GNSS hardware was active prior to it launching - it only knows when it received the first location fix since launch, when GNSS is started, and when it's stopped - these events:
https://developer.android.com/reference/android/location/GnssStatus.Callback

jidanni, 2020-04-29:
All I know is the [competition](https://play.google.com/store/apps/details?id=com.chartcross.gpstest) has a "First fix time",
[image]
The idea is the user would like to know "am I looking at figures from a 'just started' GPS, or a 'well warmed up' GPS?" even if not entirely accurate. As far as (a) vs. (b), "well something is better than nothing."

barbeau, 2020-04-29:
GPSTest has TTFF, which is time-to-first-fix, or the duration between when the app was started and the when the first GNSS fix was acquired. The above screenshot is just this value converted to a clock time.

So, if you start the app at 12:00, and it gets a fix at 12:05, GPSTest will show "5 minutes", while the above app would show "12:05".

jidanni, 2020-04-29:
"24:14" can't be HH:MM as HH only go 00-23. So You mean the above app would show 05:00.
Anyways I'm glad you have TTFF. But I would be sure it appears in the first photograph (daylight white screen) there on Google Play. (Plus some other blank fields.)

jidanni, 2020-04-30:
Oops, oops, oops. I meant that I want the current time since the current fix that I am using started. For instance, I launch the app and it takes a half a minute to finally get a fix. Okay, now  start this counter that I want Counting.

closed 2021-10-02 by stale[bot]