barbeau/gpstest #404: Logging resets on a portrait/landscape mode change

Fuzl, 2020-05-07:
**Summary:** 

A new log file is created on each portrait/landscape mode change

**Steps to reproduce:** 

- Launch app with file logging enabled. Go into "Accuracy" view. 
- Change from portrait to landscape
- An information bubble states a new logging file is created

**Expected behavior:** 

No new logging file on portrait/lanscape change

**Observed behavior:** 

**Device and Android version:** 

Pixel 4 device with latest 5th mai security update on an Android 10.
There is no Google Play Services app in my app lists :(  I guess they renamed it but could not find to what. So I cannot show the version.

**Screenshots:** 

[image]

barbeau, 2020-05-08:
@Fuzl Thanks for reporting this! I'll take a look.

barbeau, 2020-05-08:
Work-in-progress PR at https://github.com/barbeau/gpstest/pull/405.

barbeau, 2020-10-20:
Looking at the master branch and latest release v3.8.4, logs are overwritten on a Samsung Galaxy S8+ w/ Android 9 instead of being appended to. Re-opening...

commit af19685c (2020-05-27): fix: Re-use log file on orientation change (#405)

Fixes #404
commit 4e7a653a (2020-10-20): fix: Append to existing log on orientation change

Fixes #404.

I'm not sure how this worked originally when I tested it? Maybe I saw flushes of the buffer and thought they were new contents after rotation. At any rate, passing the boolean "true" into the constructor seems to work now.
closed 2020-10-20 by barbeau