barbeau/gpstest #406: Android 11 Compatibility Problem

wforw, 2020-05-19:
**Summary:** 

On Android 11 device, this app cannot show the satellites information

**Steps to reproduce:** 

Installed on Android 11

**Expected behavior:** 

Shows normal

**Observed behavior:** 

Without satellites information

**Device and Android version:** 

 Android 11

notifyDataSetChanged will  throw "Only the original thread that created a view hierarchy can touch its views."
setValue will throw "Cannot invoke setValue on a background thread"

barbeau, 2020-05-19:
@wforw Thanks for flagging this! Are you seeing this on the Android 11 emulator or a real device? If a real device, which make and model and which preview do you have loaded?

I can reproduce on the Android 11 emulator.

Here are the stack traces I'm seeing:

~~~
2020-05-19 10:53:01.514 7946-7978/com.android.gpstest W/Binder: Caught a RuntimeException from the binder stub implementation.
    android.view.ViewRootImpl$CalledFromWrongThreadException: Only the original thread that created a view hierarchy can touch its views.
        at android.view.ViewRootImpl.checkThread(ViewRootImpl.java:8705)
        at android.view.ViewRootImpl.requestLayout(ViewRootImpl.java:1570)
        at android.view.View.requestLayout(View.java:25286)
        at android.view.View.requestLayout(View.java:25286)
        at android.view.View.requestLayout(View.java:25286)
        at android.view.View.requestLayout(View.java:25286)
        at android.view.View.requestLayout(View.java:25286)
        at android.view.View.requestLayout(View.java:25286)
        at android.view.View.requestLayout(View.java:25286)
        at androidx.drawerlayout.widget.DrawerLayout.requestLayout(DrawerLayout.java:1303)
        at android.view.View.requestLayout(View.java:25286)
        at android.view.View.requestLayout(View.java:25286)
        at android.widget.RelativeLayout.requestLayout(RelativeLayout.java:380)
        at android.view.View.requestLayout(View.java:25286)
        at android.view.View.requestLayout(View.java:25286)
        at android.widget.RelativeLayout.requestLayout(RelativeLayout.java:380)
        at android.view.View.requestLayout(View.java:25286)
        at android.widget.RelativeLayout.requestLayout(RelativeLayout.java:380)
        at android.view.View.requestLayout(View.java:25286)
        at androidx.core.widget.NestedScrollView.requestLayout(NestedScrollView.java:1782)
        at android.view.View.requestLayout(View.java:25286)
        at android.view.View.requestLayout(View.java:25286)
        at android.view.View.requestLayout(View.java:25286)
        at android.view.View.requestLayout(View.java:25286)
        at android.widget.TableLayout.requestLayout(TableLayout.java:228)
        at android.view.View.requestLayout(View.java:25286)
        at android.view.View.requestLayout(View.java:25286)
        at android.widget.TextView.checkForRelayout(TextView.java:9722)
        at android.widget.TextView.setText(TextView.java:6298)
        at android.widget.TextView.setText(TextView.java:6126)
        at android.widget.TextView.setText(TextView.java:6078)
        at com.android.gpstest.GpsStatusFragment.updateFixTime(GpsStatusFragment.java:288)
        at com.android.gpstest.GpsStatusFragment.updateGnssStatus(GpsStatusFragment.java:579)
        at com.android.gpstest.GpsStatusFragment.onSatelliteStatusChanged(GpsStatusFragment.java:517)
        at com.android.gpstest.GpsTestActivity$2.onSatelliteStatusChanged(GpsTestActivity.java:1004)
        at android.location.LocationManager$GnssStatusListenerManager$GnssStatusListener.lambda$onSvStatusChanged$1(LocationManager.java:2986)
        at android.location.-$$Lambda$LocationManager$GnssStatusListenerManager$GnssStatusListener$4EPi22o4xuVnpNhFHnDvebH4TG8.accept(Unknown Source:4)
        at android.location.AbstractListenerManager$Registration.accept(AbstractListenerManager.java:81)
        at android.location.AbstractListenerManager$Registration.lambda$XpiThbVaDDpOnFWIkrt38Bf4yx0(Unknown Source:0)
        at android.location.-$$Lambda$AbstractListenerManager$Registration$XpiThbVaDDpOnFWIkrt38Bf4yx0.accept(Unknown Source:4)
        at com.android.internal.util.function.pooled.PooledLambdaImpl.doInvoke(PooledLambdaImpl.java:278)
        at com.android.internal.util.function.pooled.PooledLambdaImpl.invoke(PooledLambdaImpl.java:201)
        at com.android.internal.util.function.pooled.OmniFunction.run(OmniFunction.java:97)
        at android.location.-$$Lambda$_14QHG018Z6p13d3hzJuGTWnNeo.execute(Unknown Source:0)
        at android.location.AbstractListenerManager$Registration.execute(AbstractListenerManager.java:67)
        at android.location.AbstractListenerManager$Registration.access$200(AbstractListenerManager.java:43)
        at android.location.AbstractListenerManager.execute(AbstractListenerManager.java:206)
        at android.location.LocationManager$GnssStatusListenerManager$GnssStatusListener.onSvStatusChanged(LocationManager.java:2986)
        at android.location.IGnssStatusListener$Stub.onTransact(IGnssStatusListener.java:144)
        at android.os.Binder.execTransactInternal(Binder.java:1159)
        at android.os.Binder.execTransact(Binder.java:1123)
~~~

...and:

~~~
8/com.android.gpstest W/Binder: Caught a RuntimeException from the binder stub implementation.
    java.lang.IllegalStateException: Cannot invoke setValue on a background thread
        at androidx.lifecycle.LiveData.assertMainThread(LiveData.java:461)
        at androidx.lifecycle.LiveData.setValue(LiveData.java:304)
        at androidx.lifecycle.MutableLiveData.setValue(MutableLiveData.java:50)
        at com.android.gpstest.DeviceInfoViewModel.reset(DeviceInfoViewModel.java:276)
        at com.android.gpstest.GpsStatusFragment.updateGnssStatus(GpsStatusFragment.java:592)
        at com.android.gpstest.GpsStatusFragment.onSatelliteStatusChanged(GpsStatusFragment.java:517)
        at com.android.gpstest.GpsTestActivity$2.onSatelliteStatusChanged(GpsTestActivity.java:1004)
[... cut ...]
        at android.location.AbstractListenerManager$Registration.access$200(AbstractListenerManager.java:43)
        at android.location.AbstractListenerManager.execute(AbstractListenerManager.java:206)
        at android.location.LocationManager$GnssStatusListenerManager$GnssStatusListener.onSvStatusChanged(LocationManager.java:2986)
        at android.location.IGnssStatusListener$Stub.onTransact(IGnssStatusListener.java:144)
        at android.os.Binder.execTransactInternal(Binder.java:1159)
        at android.os.Binder.execTransact(Binder.java:1123)
~~~

barbeau, 2020-05-19:
I think this is a bug with Android R and/or R emulator, as these callbacks shouldn't be happening on a background thread. I've opened an issue on AOSP here:
https://issuetracker.google.com/issues/157059134

wforw, 2020-05-20:
I use a real device to test.  But sorry, I cannot provide the model and push the commits for company's security policy.

You can use the postValue to replace the setValue and surround the notifyDataSetChanged by RunOnUIThread.  That may be helpful

barbeau, 2020-07-02:
This has been marked as fixed by the Android team and it will become available in a future Android release:
https://issuetracker.google.com/issues/157059134#comment5

closed 2020-07-02 by barbeau