barbeau/gpstest #407: SBAS n.125

Giolly90, 2020-05-21:
I spotted on a Samsung Galaxy A6+ this SBAS satellite number 125. What is it?
[image]

barbeau, 2020-05-21:
@Giolly90 Interesting, thanks for sharing! Looks like this is SDCM, which is Russia's SBAS system:
http://mgex.igs.org/IGS_MGEX_Status_SBAS.php

See also:
https://www.gpsworld.com/gnss-systemaugmentation-assistancenewsrussian-sbas-satellite-move-13114/

I'll work on adding this to the app so it shows the proper flag.

Giolly90, 2020-05-21:
Thank you for the commitment to this fantastic app!
To add more details, I saw this sbas in northern Italy yesterday for a couple of minutes and then it was gone

commit 5373ddbb (2020-05-27): feat: Add SNAS and SDCM satellite IDs for SBAS

Based on http://mgex.igs.org/IGS_MGEX_Status_SBAS.php

Closes #407
closed 2020-05-27 by barbeau